thingsboard/thingsboard-gateway #894: [HELP] Identify the configuration files changes and auto update running connectors

STYTY, 2022-08-04:
**Describe the issue**
If I update the tb_gateway.yaml and the configuration files (*.json) from connectors (in /etc/thingsboard-gateway/config) remotely to gateway config folder, can tb gateway Identify file changes and auto update running connectors without using gateway rpc method like gateway_restart

**Configuration** 
[image]

**Versions (please complete the following information):**
 - OS: centos7
 - Thingsboard IoT Gateway version 3.1
 - Python version 3.9

samson0v, 2022-08-04:
Hi @STYTY, unfortunately no.

STYTY, 2022-08-04:
@samson0v Thank you very much!

closed 2022-08-04 by STYTY