thingsboard/thingsboard-gateway #897: [HELP] Error "curl: (56) Recv failure: Connection reset by peer" from outside docker container

technorabilia, 2022-08-06:
**Describe the issue**
I get `curl: (56) Recv failure: Connection reset by peer` when I try to curl from outside the gateway docker container.
Running the curl command from inside the docker container works as expected.

```
simon@xyz:~/Projects/tb-gateway$ cat docker-compose.yml 
version: '3.3'
services:
    tb-gateway:
        ports:
            - 5000:5000
        volumes:
            - ./logs:/thingsboard_gateway/logs
            - ./extensions:/thingsboard_gateway/extensions
            - ./config:/thingsboard_gateway/config
        container_name: tb-gateway
        restart: always
        image: thingsboard/tb-gateway
simon@xyz:~/Projects/tb-gateway$ curl -H "Content-Type: application/json" -X POST -d '{"sensorName": "SN-001", "sensorModel": "simon"}' http://127.0.0.1:5000/my_devices
curl: (56) Recv failure: Connection reset by peer
simon@xyz:~/Projects/tb-gateway$ docker exec -it 04f sh
# curl -H "Content-Type: application/json" -X POST -d '{"sensorName": "SN-001", "sensorModel": "simon"}' http://127.0.0.1:5000/my_devices
# 
```
**Configuration** (Attach your configuration file)
Just the default stuff for testing purposes.
```
thingsboard:
  host: 192.168.0.73
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: <removed>
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: REST Connector
    type: rest
    configuration: rest.json
```

```
{
  "host": "127.0.0.1",
  "port": "5000",
  "SSL": false,
  "security": {
    "type": "anonymous"
  },
  "mapping": [
    {
      "endpoint": "/my_devices",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "${sensorName}",
        "deviceTypeExpression": "${sensorType}",
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "string",
            "key": "certificateNumber",
            "value": "${certificateNumber}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "endpoint": "/anon1",
      "HTTPMethods": [
        "GET",
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "Device 2",
        "deviceTypeExpression": "default",
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "Model2"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
    {
      "endpoint": "/anon2",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "converter": {
        "type": "custom",
        "deviceNameExpression": "SuperAnonDevice",
        "deviceTypeExpression": "default",
        "extension": "CustomRestUplinkConverter",
        "extension-config": [
          {
            "key": "Totaliser",
            "datatype": "float",
            "fromByte": 0,
            "toByte": 4,
            "byteorder": "big",
            "signed": true,
            "multiplier": 1
          }
        ]
      }
    }
  ],
  "attributeRequests": [
    {
      "endpoint": "/sharedAttributes",
      "type": "shared",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "timeout": 10.0,
      "deviceNameExpression": "${deviceName}",
      "attributeNameExpression": "${attribute}${attribute1}"
    }
  ],
  "attributeUpdates": [
    {
      "HTTPMethod": "POST",
      "SSLVerify": false,
      "httpHeaders": {
        "CONTENT-TYPE": "application/json"
      },
      "security": {
        "type": "anonymous"
      },
      "timeout": 0.5,
      "tries": 3,
      "allowRedirects": true,
      "deviceNameFilter": "SN.*",
      "attributeFilter": ".*",
      "requestUrlExpression": "http://127.0.0.1:5001/",
      "valueExpression": "{\"deviceName\":\"${deviceName}\",\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "echo",
      "requestUrlExpression": "http://127.0.0.1:5001/${deviceName}",
      "responseTimeout": 1,
      "HTTPMethod": "GET",
      "valueExpression": "${params}",
      "timeout": 10.0,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json"
      },
      "security": {
        "type": "anonymous"
      }
    },
    {
      "deviceNameFilter": "SN.*",
      "methodFilter": "post_attributes",
      "requestUrlExpression": "http://127.0.0.1:5000/my_devices",
      "responseTimeout": 1,
      "HTTPMethod": "POST",
      "valueExpression": "{\"sensorName\":\"${deviceName}\", \"sensorModel\":\"${params.sensorModel}\", \"certificateNumber\":\"${params.certificateNumber}\", \"temp\":\"${params.temp}\", \"hum\":\"${params.hum}\"}",
      "timeout": 10.0,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json"
      },
      "security": {
        "type": "anonymous"
      }
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "HTTPMethod": "POST",
      "valueExpression": "${params}",
      "httpHeaders": {
        "Content-Type": "application/json"
      },
      "security": {
        "type": "anonymous"
      }
    }
  ]
}
```

**Connector name (If you need help with some connector/converter):**
REST Connector.

**Error traceback (If it was raised):**
Not applicable.

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Docker: 20.10.17
 - Thingsboard IoT Gateway version: 3.1
 - Python version: not applicable

samson0v, 2022-08-08:
Hi @technorabilia, it happens because you are using localhost for the rest connector, try to use the following config:
```
{
  "host": "0.0.0.0",
  "port": "5000",
  "SSL": false,
  "security": {
```

technorabilia, 2022-08-08:
Thanks that was indeed the cause.

closed 2022-08-08 by technorabilia

Hellal1997, 2024-07-31:
curl localhost:9090

connection reset by peer how could i solve this issue

could anaybody guide me