thingsboard/thingsboard-gateway #898: [HELP] Thingsboard subscribe via IoT gateway does not work

lvojp, 2022-08-09:
Hi there, nice to meet you.

I am trying to communicate with Thingsboard using the MQTT connector of the IoT gateway, following the tutorial in the URL below.

[MQTT Connector Configuration | ThingsBoard IoT Gateway](https://thingsboard.io/docs/iot-gateway/config/mqtt/)

I am able to create devices and update data (humidity, temperature) on Thingsboard using Mosquitto_pub via IoT gateway, but I am not able to subscribe items in Firmware.

If I can't do this, I can't do bi-directional communication through the IoT Gateway, and I'm having trouble.

---

# Environment

Host 1.
Thingsboard is running on Host 1.
The OS is Ubuntu 20.04.3 LTS.

Thingsboard uses the following image from Docker
"thingsboard/tb-postgres:3.3.2"

Host 2.
Host 2 is running IoT Gateway and Mosquitto.
The operating system is Ubuntu 18.04.6 LTS.

The IoT Gateway uses the following image from Docker.
thingsboard/tb-gateway:3.1

Mosquitto_sub version is 1.4.15.

--- 

# About the problem

Specifically, when I execute the command mosquitto_sub, I get the following situation.

```
mosquitto_sub -h 127.0.0.1 -t /sensor/SN-002/FirmwareVersion -d -v
Client mosqsub|7613-lapis sending CONNECT
Client mosqsub|7613-lapis received CONNACK
Client mosqsub|7613-lapis sending SUBSCRIBE (Mid: 1, Topic: /sensor/SN-002/FirmwareVersion, QoS: 0)
```

In this situation, run the following command to update the device data once and then run

`mosquitto_pub -h 192.168.11.22 -p 1883 -t "/sensor/SN-001/data" -m '{"sensorType": "Thermometer", "sensorModel": "T1000", "temp": 100, "hum": 60}''`

When I change the FirmwareVersion of the SN-001 on the Thingsboard, I get the following error in the IoT Gateway.

`tb-gateway | ""2022-08-09 07:32:39"" - |ERROR| - [mqtt_connector.py] - mqtt_connector - on_attributes_update - 610 - Cannot find attributeName by filter in message with data: {'device': 'SN-002', 'data': {'FirmwareVersion': '1'}}"`

Given the situation, it seems that the communication is working from the Thingsboard to the IoT Gateway, but it seems that either the IoT Gateway or a problem with Mosquitto_sub is preventing it from subscribing.

Can anyone help me with this problem?

Your help would be greatly appreciated.
Thank you in advance.

samson0v, 2022-08-09:
Hi @lvojp, send, please, your config files.

Thanks for your interest in ThingsBoard IoT Gateway!

lvojp, 2022-08-09:
Thank you for your reply. @samson0v 
I have prepared a config file.

Sorry, the host IP and access token in tb_gateway.yaml have been removed for security reasons.

[mqtt.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/9288721/mqtt.json.txt)
[tb_gateway.yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/9288723/tb_gateway.yaml.txt)

samson0v, 2022-08-09:
@lvojp, you have to use following config:
```
"attributeUpdates": [
    {
      "retain": false,
      "deviceNameFilter": "SN.*",
      "attributeFilter": "FirmwareVersion",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
  ],
```

**Explanation:**
You want to update `FirmwareVersion` but in the config `attributeFilter` parameter you are using `uploadFrequency`, so connector can't find the right attrUpdate.
Also, make sure to configure the right `topicExpression` where the msg will send with a new attribute value.

lvojp, 2022-08-10:
@samson0v 

I have tried the method you gave me and I can confirm that it works fine.

I sincerely appreciate your help!
Thank you !

closed 2022-08-10 by lvojp