thingsboard/thingsboard-gateway #900: [HELP] How to perform data transformation in TB Gateway?

technorabilia, 2022-08-09:
**Describe the issue**
We would like to convert the data telemetry (see below) which is in base64 format to hex format.

In ThingsBoard PE this can be done in the converter / function decoder by defining a helper function.

How is this done using the TB Gateway?

[image]

**Configuration** (Attach your configuration file)
```
{
  "host": "0.0.0.0",
  "port": "7070",
  "SSL": false,
  "security": {
    "type": "anonymous"
  },
  "mapping": [
    {
      "endpoint": "/chirpstack",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "${deviceName}",
        "deviceTypeExpression": "${applicationName}",
        "attributes": [
          {
            "type": "string",
            "key": "devEUI",
            "value": "${devEUI}"
          }
        ],
        "timeseries": [
          {
            "type": "string",
            "key": "data",
            "value": "${data}"
          },
         {
            "type": "string",
            "key": "rxInfo[0].gatewayID",
            "value": "${rxInfo[0].gatewayID}"
          },
          {
            "type": "string",
            "key": "txInfo.frequency",
            "value": "${txInfo.frequency}"
          }
        ]
      }
    },
<snip>
```

**Connector name (If you need help with some connector/converter):**
REST Connector.

**Error traceback (If it was raised):**
Not applicable.

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Docker: 20.10.17
 - Thingsboard IoT Gateway version: 3.1
 - Python version: not applicable

samson0v, 2022-08-10:
Hi @technorabilia, you have to use custom converters. There is [official instruction](https://thingsboard.io/docs/iot-gateway/custom/#custom-converter-methods-reference) on how to write your custom converter.

samson0v, 2022-08-10:
@technorabilia, also, you can parse the data in the rule chain it will be more simpler.

technorabilia, 2022-08-10:
Thanks @samson0v !