thingsboard/thingsboard-gateway #901: [HELP] Efficient implementation of a custom REST connector by reusing existing code?

technorabilia, 2022-08-10:
**Describe the issue**
The REST Connection works for us as expected but we would like some minor modifications to do some data transformation and add maybe some helper functions for it.

See also #900 for the question I asked earlier (we don't want to use the rule chain for this and chose to create a custom connector).

Pseudo code (added the base64.b64decode()):
```
        "timeseries": [
          {
            "type": "string",
            "key": "data",
            "value": "base64.b64decode(${data})"
          },
```

I am aware of the Custom IoT Gateway Connector that can be found [here](https://thingsboard.io/docs/iot-gateway/custom/).

In my opinion, only a small adjustment is needed and I don't really want to replicate the entire existing REST connector functionality because it already works for 99% as required.

What's the best approach here? Could use some tips...

**Configuration** (Attach your configuration file)
Not applicable.

**Connector name (If you need help with some connector/converter):**
Custom REST Connector.

**Error traceback (If it was raised):**
Not applicable.

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Docker: 20.10.17
 - Thingsboard IoT Gateway version: 3.1
 - Python version: not applicable

samson0v, 2022-08-12:
Hi @technorabilia, I think that you have to implement only custom converter, not all connector.

technorabilia, 2022-08-13:
Thanks @samson0v , makes sense. I will give it a try.

closed 2022-08-13 by technorabilia