thingsboard/thingsboard-gateway #902: [HELP] Set "Allow create devices or assets" to NO in TB Gateway?

technorabilia, 2022-08-17:
**Describe the issue**
In a TB PE Integration you can set "Allow create devices or assets" to NO so devices are not automatically created.

Is this also possible using the TB CE Gateway?

I cannot seem to find a setting for this?

samson0v, 2022-08-18:
Hi @technorabilia, it is not possible in Gateway.

technorabilia, 2022-08-19:
Thanks, if you create a data converter in TB PE, the default decoder function is like the one below.

Here you can switch between `deviceName` and `deviceType` or `assetName` and `assetType`.

We have similar functionality that dynamically switches between device and asset and process the attributes and telemetry accordingly.

Is this also possible in the TB CE Gateway?

I guess I am looking for something like `assetNameExpression` and `assetTypeExpression` similar to `deviceNameExpression` and `deviceTypeExpression`?

I've searched the code and didn't find any references to asset, but I'm asking here to make sure...

```
// Decode an uplink message from a buffer
// payload - array of bytes
// metadata - key/value object

/** Decoder **/

// decode payload to string
var payloadStr = decodeToString(payload);

// decode payload to JSON
// var data = decodeToJson(payload);

var deviceName = 'Device A';
var deviceType = 'thermostat';
var customerName = 'customer';
var groupName = 'thermostat devices';
// use assetName and assetType instead of deviceName and deviceType
// to automatically create assets instead of devices.
// var assetName = 'Asset A';
// var assetType = 'building';

// Result object with device/asset attributes/telemetry data
var result = {
// Use deviceName and deviceType or assetName and assetType, but not both.
   deviceName: deviceName,
   deviceType: deviceType,
// assetName: assetName,
// assetType: assetType,
   customerName: customerName,
   groupName: groupName,
   attributes: {
       model: 'Model A',
       serialNumber: 'SN111',
       integrationName: metadata['integrationName']
   },
   telemetry: {
       temperature: 42,
       humidity: 80,
       rawData: payloadStr
   }
};

/** Helper functions **/

function decodeToString(payload) {
   return String.fromCharCode.apply(String, payload);
}

function decodeToJson(payload) {
   // covert payload to string.
   var str = decodeToString(payload);

   // parse string to JSON
   var data = JSON.parse(str);
   return data;
}

return result;
```

samson0v, 2022-08-22:
Hi @technorabilia, no, Gateway is working only with devices.
For working with assets and other entities you can use our [rest client](https://github.com/thingsboard/thingsboard-python-rest-client).

samson0v, 2022-08-26:
Hi @technorabilia, can we close the issue?

closed 2022-08-26 by technorabilia