thingsboard/thingsboard-gateway #903: [HELP]

Lenteguppie, 2022-08-25:
**Describe the issue**
I cannot create a custom REST converter for docker. When I want to create a custom REST converter like with the converter configuration based on the documentation. I get an error when I an sending a curl request that a 'NoneType' is not callable.

I use the gateway with docker-compose. I put the converter in the extentions folder which is generated by docker-compose. 

Do you have any idea to fix this error? Thanks in advance!  

**Configuration** (Attach your configuration file)
***rest.json***
```json
{
    "host": "0.0.0.0",
    "port": "5000",
    "SSL": false,
    "mapping": [
        {
            "endpoint": "/anon2",
            "HTTPMethods": ["POST"],
            "security": {
                "type": "anonymous"
            },
            "converter": {
                "type": "custom",
                "deviceNameExpression": "SuperAnonDevice",
                "deviceTypeExpression": "default",
                "extension": "CustomRestUplinkConverter",
                "extension-config": [
                    {
                        "key": "Totaliser",
                        "datatype": "float",
                        "fromByte": 0,
                        "toByte": 4,
                        "byteorder": "big",
                        "signed": true,
                        "multiplier": 1
                    }
                ]
            }
        }
    ],
    "attributeRequests": [
        {
            "endpoint": "/sharedAttributes",
            "type": "shared",
            "HTTPMethods": ["POST"],
            "security": {
                "type": "anonymous"
            },
            "timeout": 10.0,
            "deviceNameExpression": "${deviceName}",
            "attributeNameExpression": "${attribute}${attribute1}"
        }
    ],
    "attributeUpdates": [
        {
            "HTTPMethod": "POST",
            "SSLVerify": false,
            "httpHeaders": {
                "CONTENT-TYPE": "application/json"
            },
            "security": {
                "type": "anonymous"
            },
            "timeout": 0.5,
            "tries": 3,
            "allowRedirects": true,
            "deviceNameFilter": "SN.*",
            "attributeFilter": ".*",
            "requestUrlExpression": "http://127.0.0.1:5001/",
            "valueExpression": "{\"deviceName\":\"${deviceName}\",\"${attributeKey}\":\"${attributeValue}\"}"
        }
    ],
    "serverSideRpc": [
        {
            "deviceNameFilter": ".*",
            "methodFilter": "echo",
            "requestUrlExpression": "http://127.0.0.1:5001/${deviceName}",
            "responseTimeout": 1,
            "HTTPMethod": "GET",
            "valueExpression": "${params}",
            "timeout": 10.0,
            "tries": 3,
            "httpHeaders": {
                "Content-Type": "application/json"
            },
            "security": {
                "type": "anonymous"
            }
        },
        {
            "deviceNameFilter": "SN.*",
            "methodFilter": "post_attributes",
            "requestUrlExpression": "http://127.0.0.1:5000/my_devices",
            "responseTimeout": 1,
            "HTTPMethod": "POST",
            "valueExpression": "{\"sensorName\":\"${deviceName}\", \"sensorModel\":\"${params.sensorModel}\", \"certificateNumber\":\"${params.certificateNumber}\", \"temp\":\"${params.temp}\", \"hum\":\"${params.hum}\"}",
            "timeout": 10.0,
            "tries": 3,
            "httpHeaders": {
                "Content-Type": "application/json"
            },
            "security": {
                "type": "anonymous"
            }
        },
        {
            "deviceNameFilter": ".*",
            "methodFilter": "no-reply",
            "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
            "HTTPMethod": "POST",
            "valueExpression": "${params}",
            "httpHeaders": {
                "Content-Type": "application/json"
            },
            "security": {
                "type": "anonymous"
            }
        }
    ]
}
```
***CustomRestUplinkConverter.py***
``` python
#     Copyright 2022. ThingsBoard
#
#     Licensed under the Apache License, Version 2.0 (the "License");
#     you may not use this file except in compliance with the License.
#     You may obtain a copy of the License at
#
#         http://www.apache.org/licenses/LICENSE-2.0
#
#     Unless required by applicable law or agreed to in writing, software
#     distributed under the License is distributed on an "AS IS" BASIS,
#     WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
#     See the License for the specific language governing permissions and
#     limitations under the License.

from thingsboard_gateway.connectors.converter import Converter, log

class CustomRestUplinkConverter(Converter):
    def __init__(self, config):
        log.info('initializing')
        self.__config = config
        log.info('initialized')
        self.result_dict = {
            'deviceName': config.get('name', 'CustomRestDevice'),
            'deviceType': config.get('deviceType', 'default'),
            'attributes': [],
            'telemetry': []
        }

    def convert(self, config, data: bytes):
        log.info('Converting')
        keys = ['attributes', 'telemetry']
        for key in keys:
            self.result_dict[key] = []
            if self.__config.get(key) is not None:
                for config_object in self.__config.get(key):
                    data_to_convert = data
                    if config_object.get('untilDelimiter') is not None:
                        data_to_convert = data.split(config_object.get(
                            'untilDelimiter').encode('UTF-8'))[0]
                    if config_object.get('fromDelimiter') is not None:
                        data_to_convert = data.split(config_object.get(
                            'fromDelimiter').encode('UTF-8'))[1]
                    if config_object.get('toByte') is not None:
                        to_byte = config_object.get('toByte')
                        if to_byte == -1:
                            to_byte = len(data) - 1
                        data_to_convert = data_to_convert[:to_byte]
                    if config_object.get('fromByte') is not None:
                        from_byte = config_object.get('fromByte')
                        data_to_convert = data_to_convert[from_byte:]
                    converted_data = {
                        config_object['key']: data_to_convert.decode('UTF-8')}
                    self.result_dict[key].append(converted_data)
        log.debug("Converted data: %s", self.result_dict)
        return self.result_dict

```

***docker-compose.yaml***
``` yaml
version: '3.0'
services:
  tb-gateway:
    restart: always
    image: "thingsboard/tb-gateway:latest"
    # Ports (Required for REST connector)
    ports:
      - "5000:5000"
    volumes:
      - .tb-gateway/config:/thingsboard_gateway/config
      - .tb-gateway/logs:/thingsboard_gateway/logs
      - .tb-gateway/extensions:/thingsboard_gateway/extensions
```

**Connector name (If you need help with some connector/converter):**
Rest connector and Rest converter

**Error traceback (If it was raised):**
```bash
tb-gateway_1  | ""2022-08-25 11:30:25" - |INFO| - [rest_connector.py] - rest_connector - __call__ - 477 - CONVERTER CONFIG: {'type': 'custom', 'deviceNameExpression': 'SuperAnonDevice', 'deviceTypeExpression': 'default', 'extension': 'CustomRestUplinkConverter', 'extension-config': [{'key': 'Totaliser', 'datatype': 'float', 'fromByte': 0, 'toByte': 4, 'byteorder': 'big', 'signed': True, 'multiplier': 1}]}"
tb-gateway_1  | ""2022-08-25 11:30:25" - |ERROR| - [rest_connector.py] - rest_connector - __call__ - 488 - Error while post to anonymous handler: 'NoneType' object is not callable"
tb-gateway_1  | Traceback (most recent call last):
tb-gateway_1  |   File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.1-py3.9.egg/thingsboard_gateway/connectors/rest/rest_connector.py", line 478, in __call__
tb-gateway_1  |     converter = self.endpoint['converter'](endpoint_config['converter'])
tb-gateway_1  | TypeError: 'NoneType' object is not callable

```

**Versions (please complete the following information):**
 - OS: docker-compose
 - Thingsboard IoT Gateway version 3.1
 - Python version: The version embedded in the docker container

samson0v, 2022-08-26:
Hi @Lenteguppie, we will investigate your problem and let you know about the result.

Thanks for your interest in ThingsBoard IoT Gateway!

Lenteguppie, 2022-08-29:
Hi @samson0v, 

Thank you for your response. Do you have any updates on the progress of the investigation?

samson0v, 2022-08-29:
Hi @Lenteguppie, you have to add one more volume:
`      - .tb-gateway/extensions:/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.1-py3.9.egg/thingsboard_gateway/extensions`

Lenteguppie, 2022-08-29:
This solution works, thank you!

closed 2022-08-29 by Lenteguppie

Lenteguppie, 2022-08-31:
I am able to create a custom converter, but I still have a question.

Why do I have to add the extra volume which is pointing to the extensions folder? I actually expected one volume pointing to that directory. 

Is it planned to make the extensions directory one docker volume instead of two?

Thanks in advance.