thingsboard/thingsboard-gateway #904: No Data from Kepware OPC-UA Server

fredguo2010, 2022-08-26:
I have configured the tb gateway with docker and OPC-UA server with Kepware Enterprise, and successfully connected based on logs.
[image]
However, there is no data push to the tb platform
[image]
My opc-ua.json file looks as this
[image]
Kepware interface looks as this
[image]

Thingsboard Gateway Version: 3.1

Thanks a lot for anyone who can provide help.

samson0v, 2022-08-29:
Hi @fredguo2010, please, send the screenshot log using DEBUG level (in logs.conf file replace all INFO to DEBUG).
And I think you set the wrong path to your values (from the screen I can't see ns and i).

The next thing,
[image]
You write the wrong delimiter `,` have to be `;`

fredguo2010, 2022-08-29:
@samson0v  Thanks a lot for your quick response.
I have reconfigured with ns and s after reviewing the Kepware Server and Quick Client, see below 2 pictures:
[image]
[image]

However still there is no opc data from tb gateway
[image]

See below for debug, there is only modbus connector info
[image]

samson0v, 2022-08-29:
@fredguo2010 please, send the screenshot log using DEBUG level **(in logs.conf file replace all INFO to DEBUG)**.
For now, I see only INFO-level logs.
[image]

Also, disable the Modbus connector temporarily.

fredguo2010, 2022-08-30:
@samson0v  DEBUG level included
[image]

samson0v, 2022-08-30:
Hi @fredguo2010, from the logs, I see that connector can't find the devices (misconfiguration).
Try to use `showMap: true` parameter to see how the connector iterate through the nodes tree.

fredguo2010, 2022-08-30:
@samson0v 
[image]
Here is what i got after config showMap with true, might the path of device node pattern should be updated, please advise, thank you

samson0v, 2022-08-31:
Hi @fredguo2010, from the logs, you have to use `Root.Objects.Server.SIM1...`
[image]

fredguo2010, 2022-08-31:
@samson0v 
Still no mapping after reconfig the node, please advise if i misconfigured
[image]

samson0v, 2022-08-31:
@fredguo2010, maybe it is in the higher level, try `Root.SIM1....`

fredguo2010, 2022-08-31:
@samson0v the Device1 is not even searched if moving to upper level...
[image]

samson0v, 2022-10-19:
@fredguo2010 any updates?

mica83, 2022-11-21:
Hello, @fredguo2010 any updates? we have the some problem!.

Thanks

samson0v, 2022-11-22:
@SimoneG97, thanks for the information, we will add a warning to the documentation.