thingsboard/thingsboard-gateway #905: [BUG]Modbus rtu (serial) multiple device connection issue

Maksym2022, 2022-08-30:
**Describe the bug**
One Modbus slave is polled correctly. Interrogation of two or more Modbus devices is performed incorrectly (violation of the rules for polling devices of the Modbus protocol)! The Modbus RTU protocol requires only one master device at a time. The order of polling devices should be as follows: send a request to the device - we are waiting for a response, receive a response - we poll the next device. While the response from the device has not yet been received, it is necessary to wait for a timeout!

In practice, the gateway works as follows (for Modbus RTU connector):
1. For each declared device, a separate daemon thread is created.
2. These separate threads try to poll each of their devices independently of each other through one port tmyUSB0!
3. Requests from different threads leave the port tmyUSB0 one after another without pause! This is a common mistake when different threads use the same resource at the same time. In this case, the port ttuUSB0.
4. To use Modubus RTU, you need to add thread blocking (as a mutex). The first thread takes the mutex, polls its device, gets a response, releases the mutex, goes to sleep... The next thread takes the mutex, polls its device... and so on. round.
5. For the case of Modbus TСP, parallel operation of threads is allowed. There are no problems with polling multiple devices.

**Connector name (If bug in the some connector):**
[modbus serial connector]

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [Ubuntu 22.04]
 - Thingsboard IoT Gateway version [latest on 2022.08.30]
 - Python version[3.0]

samson0v, 2022-08-30:
Hi @Maksym2022, thanks for your interest in ThingsBoard IoT Gateway! 
We will investigate this situation and let you know about the result.

Maksym2022, 2022-09-14:
I resolved this bug by modifying file modbus_connector.py from: /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/
I added mutex in "   def __process_slaves(self):" method. I tested it only in case of modbus rtu.

Check attached zip file
[modbus_connector.zip](https://github.com/thingsboard/thingsboard-gateway/files/9566945/modbus_connector.zip)

Maksym2022, 2022-09-16:
Hi, everyone!
I made some additional tests. I want tell you about my results:

1. First i added mutex at "def __process_slaves(self):" method in file "modbus_connector.py" (i wrote about it elear). I tested it with modbus rtu only. I configured gateway to communicate with two devices in file "modbus_serial.json". Then i monitored communication traffic with these devices. It work correct. Gateway requests all registers from first device, then all registers from second, then again from first. It's good traffic!

2. Second i tested modbus tcp communication! Here I was in for an unpleasant surprise. There was no data exchange with devices! Then i reverted changes (removed mutex) and test again. Requests to devices was ok!

3. I changed "modbus_connector.py" again! Before set mutex, i check connection type (if device.config.get(TYPE_PARAMETER) == 'serial':    self.lock.acquire()). And tested again! As it should be with modbus tcp only communication was ok.

4. And finaly i tryed to test bouth communication type modbus rtu and  modbus tcp at the same time! I monitored devices requests traffic. TCP requests was ok, modbus rtu requests with some errors! BUT there are no timeseries telemetry data updates on thingsboard dashboards!!!

Conclutions are: when use modbus tcp or modbus rtu separate, connectors work ok! When use it simultaniusly it doesn't work! I dont know why! 

Maybe it will be useful to separate tcp and rtu connectors into different parts: modbus_serial & modbus_tcp at python connectors folder!
Please check attached files!
[modbus_connector.zip](https://github.com/thingsboard/thingsboard-gateway/files/9581625/modbus_connector.zip)

closed 2022-10-19 by imbeacon