thingsboard/thingsboard-gateway #906: [HELP] How to properly drop a message inside a converter

raomin, 2022-09-01:
**Describe the issue**
For some reason I need to discard some MQTT in my mqtt converter.
But the `def convert(self, topic, body)` function requires a return.

If I `return None` for discarded message, I get an error in the log:
```log
""2022-09-01 12:14:10" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 668 - argument of type 'NoneType' is not iterable"
```
Similar error when I `return {}`.

What should I return to silently drop a message sent to `convert()`?

samson0v, 2022-09-02:
Hi @raomin, you have to return the default data structure, as in the following example:
[image]

raomin, 2022-09-02:
Thanks @samson0v  I changed it as you said. It is now properly dropped.

_For those wanting to copy paste the line:_
```py
dict_result = {"deviceName": None, "deviceType": None, "attributes": [], "telemetry": []}
```

raomin, 2022-09-02:
Well actually, I was to quick... It's still complaining, and even more now...
```log
""2022-09-02 11:16:30" - |ERROR| - [tb_utility.py] - tb_utility - validate_converted_data - 67 - deviceName is empty in data: {"deviceName": null, "deviceType": null, "attributes": [], "telemetry": []}"
""2022-09-02 11:16:30" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 608 - Data from MQTT Broker Connector connector is invalid."
```

samson0v, 2022-09-07:
@raomin, but you have to fill deviceName and deviceType.

raomin, 2022-09-07:
I receive a message on MQTT that is not for my platform. I just want to ignore this message...
So, even if I just want to drop the message I need to link it to a device... That would make this device active I guess.
But ok, I can link it to a dummy device...
I'll try this. Thanks.

samson0v, 2022-09-19:
Hi @raomin, any updates? Can we close this issue?

closed 2022-09-19 by raomin