thingsboard/thingsboard-gateway #912: [HELP] thingsboard-iot-gateway doesn´t start...

rungg12, 2022-09-10:
Hey!

I just made a fresh install of Thingsboard and of the Gateway. Thingsboard service is running without any problems, but the gateway just doesn´t want to start. 

I get following error: 
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'

I almost tried everything on google, but always the same error.
I hope someone can help me

BTW, I am on Ubuntu, 20.04.

devaskim, 2022-09-10:
What installation mode did you use for gateway?

rungg12, 2022-09-10:
i used the guide on the website: https://thingsboard.io/docs/iot-gateway/install/deb-installation/

devaskim, 2022-09-11:
Try to totally remove the package and then install again
```
sudo aptitude purge packagename
```

samson0v, 2022-09-12:
Hi @rungg12, try to run:
`sudo pip3 install protobuf==3.20.0`

rungg12, 2022-09-13:
@samson0v your tipp did it, @devaskim thank you too

closed 2022-09-14 by imbeacon