thingsboard/thingsboard-gateway #914: [BUG] Failed to start locally build Docker image

devaskim, 2022-09-10:
**Describe the bug**
I made this PR #913 and built Docker container to test
```
export GW_DOCKER_TAG=3.2 && \
docker build -t devaskim/tb-gateway:$GW_DOCKER_TAG docker/ && \
docker push devaskim/tb-gateway:$GW_DOCKER_TAG
```
Docker Compose configuration is [official](https://github.com/thingsboard/thingsboard-gateway/blob/master/docker/docker-compose.yml).

Here is what I got
```
admin-docker@test:~/gateway$ docker-compose up
[+] Running 1/0
 ⠿ Container gateway-tb-gateway-1  Recreated                                                                                                                                                                 0.1s
Attaching to gateway-tb-gateway-1
gateway-tb-gateway-1  | python: can't open file '//./thingsboard_gateway/tb_gateway.py': [Errno 2] No such file or directory
gateway-tb-gateway-1 exited with code 2
gateway-tb-gateway-1  | python: can't open file '//./thingsboard_gateway/tb_gateway.py': [Errno 2] No such file or directory
gateway-tb-gateway-1 exited with code 2
```
**Versions (please complete the following information):**
 - OS: [Ubuntu 
 - Thingsboard IoT Gateway version 3.1

samson0v, 2022-09-12:
Hi @devaskim, under the docker folder have to be all gateway files.
[image]

devaskim, 2022-09-12:
Thanks @samson0v Is it enough to move `Dockerfile` on up root project directory instead of copying all files to `docker` subdirectory?
It would be great to have corresponding documentation section to avoid extra questions.

samson0v, 2022-09-12:
@devaskim, we will mention it in the documentation.

devaskim, 2022-09-12:
Great, I suggest to close this ticket after docs will be added ;)

closed 2022-11-14 by samson0v

yosimasu, 2023-03-23:
Using `docker build . -f docker/Dockerfile` will build success.