thingsboard/thingsboard-gateway #916: [HELP] MQTT - regexp in topic filter

devaskim, 2022-09-13:
Broker topic structure is:
```
SN1234/
    PV_description/
    KPI_description/
SN54321/
    PV_description/
    KPI_description/
```
How to subscribe both to *PV_description* and to *KPI_description*? Thanks

samson0v, 2022-09-14:
Hi @devaskim, gateway can't handle regex in a topic so you can restructure topics, for example:
SN1234/
    some_topic1/
    some_topic2/
    PV/description/
    KPI/description/

And use the next `topicFilter`: SN1234/+/description. It will be more similar to the MQTT topic design.

devaskim, 2022-09-14:
Thanks @samson0v

samson0v, 2022-09-14:
@devaskim, got some updates for you, Gateway can handle regex in a topic. 
My bad.

devaskim, 2022-09-14:
Great, could you provide some example?

devaskim, 2022-09-14:
Great, @samson0v could you provide some example?

samson0v, 2022-09-15:
Hi @devaskim, the MQTT client can't subscribe to topics that include some regex.
So you can't use them in configuration.
Sorry for the misinformation.

samson0v, 2022-09-16:
@devaskim, can we close the issue?

devaskim, 2022-09-16:
Yes, thanks.
Also waiting for response in other tickets ;)

closed 2022-09-16 by devaskim