thingsboard/thingsboard-gateway #919: [FEATURE] Optimize docker image

solred, 2022-09-13:
At this moment the docker image is now almost 900 MB.

Probably / possibly this can be optimized?

https://hub.docker.com/r/thingsboard/tb-gateway

```
REPOSITORY                                                          TAG            IMAGE ID       CREATED        SIZE
thingsboard/tb-gateway                                              latest         67d20b264edf   3 months ago   897MB
```

samson0v, 2022-09-14:
Hi @solred, We already optimized the docker image and fixed the bug with correctly binding extensions.
[image]
So you can build a docker image using Dockerfile from the master branch.

solred, 2022-09-14:
Thanks, I will take a look at it.

closed 2022-09-14 by solred