thingsboard/thingsboard-gateway #924: [BUG] snmp connector is not polling the device

hansju-11, 2022-09-18:
I did:
1.) activate snmp connector in tb_gateway.yaml
...
  -
    name: SNMP Connector
    type: snmp
    configuration: snmp.json

....
2.) set snmp.json

{
  "devices": [
    {
      "deviceName": "NAS",
      "deviceType": "snmp",
      "ip": "192.168.101.222",
      "port": 161,
      "pollPeriod": 60,
      "community": "b-public",
          "attributes": [
        {
          "key": "ReceivedFromGet",
          "method": "get",
          "oid": "1.3.6.1.2.1.1.1.0",
          "timeout": 6
        },
          "telemetry": [
           { "key": "Temp",
          "community": "b-public",
          "method": "get",
          "oid": ".1.3.6.1.4.1.6574.1.2.0"
        },
        {
        "key": "Update-Status",
          "community": "b-public",
          "method": "get",
          "oid": ".1.3.6.1.4.1.6574.1.5.4.0",
          "timeout": 6
        }
      ],
          "attributeUpdateRequests": [
        {
          "attributeFilter": "dataToSet",
          "method": "set",
          "oid": "1.3.6.1.2.1.1.1.0",
          "timeout": 6
        }
      ],
      "serverSideRpcRequests": [
        {
          "requestFilter": "setData",
          "method": "set",
          "oid": "1.3.6.1.2.1.1.1.0"
        }
         ]
    }
  ]
}

When I start thingsboard-gateway I did not get any error in the /var/log/thingsboard-gateway/ 
If I check network communication with tcpdump host 192.168.101.222 I did not get any request from thingsboard-gateway host to snmp device.

OS: "Ubuntu 20.04.5 LTS
Thingsboard gateway 3.1.1
thingsboard 3.3.4.1

samson0v, 2022-09-19:
Hi @hansju-11, make sure that the log level is DEBUG in logs.conf file (replace all INFO to DEBUG if it is not) and send the logs.

hansju-11, 2022-09-19:
Hello @samson0v see the logfiles 
[thingsboard-gateway.tar.gz](https://github.com/thingsboard/thingsboard-gateway/files/9599909/thingsboard-gateway.tar.gz)

samson0v, 2022-10-14:
@hansju-11 from the logs files I see only MQTT Connector logs, maybe it will be better if you send a screenshot of the logs when the gateway starts to poll SNMP devices.

samson0v, 2022-10-19:
@hansju-11 any updates?

hansju-11, 2022-10-19:
Hello Vitalii,I'm on vacation. I will check it next week.Kind regardsHans-Jürgen --Diese Nachricht wurde von meinem Mobiltelefon gesendet.

hansju-11, 2022-10-26:
Hello Vitalii,

I did an wireshark trace, but no snmp connection was made. It looks like that their is no thread/task created in thingsboard to do the snmp.

1.) Can I check, if the config file has any problems

2.) is read by thingsboard

3.) if puresnmp is started

 

kind regards

Hans-Jürgen

 

 

 
 

Gesendet: Mittwoch, 19. Oktober 2022 um 14:45 Uhr

closed 2022-11-21 by imbeacon