thingsboard/thingsboard-gateway #926: [HELP] json array with request connector

wesmcpl, 2022-09-19:
I am having a little trouble getting my json data into Thingsboard. Maybe somebody could point me in the right direction.

This is what my json data looks like

**[{"name":"BN","count":154},{"name":"CC","count":22},{"name":"HD","count":48}]**

In the logs, the converted data looks like this.

""2022-09-19 15:07:44" - |ERROR| - [tb_utility.py] - tb_utility - validate_converted_data - 67 - deviceName is empty in data: **{"BN": {"deviceName": "BN", "deviceType": "default", "attributes": [], "telemetry": [{"count": "154"}]}, "CC": {"deviceName": "CC", "deviceType": "default", "attributes": [], "telemetry": [{"count": "23"}]}, "HD": {"deviceName": "HD", "deviceType": "default", "attributes": [], "telemetry": [{"count": "48"}]}, "telemetry": [], "attributes": []}**"
""2022-09-19 15:07:44" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 608 - Data from REQUEST Connector connector is invalid."

**request.json**
{
  "host": "xxx",
  "SSLVerify": true,
  "security": {
    "type": "anonymous"
  },
  "mapping": [
    {
      "url": "xxx",
      "httpMethod": "GET",
      "httpHeaders": {
        "ACCEPT": "application/json"
      },
      "allowRedirects": true,
      "timeout": 5,
      "scanPeriod": 60,
"converter": {
        "type": "json",
        "deviceNameJsonExpression": "${name}",
        "deviceTypeJsonExpression": "default",
        "attributes": [],
        "telemetry": [
          {
            "key": "count",
            "type": "int",
            "value": "${count}"
          }
        ]
      }
    }
  ]
}

Is there anything special I need to do to be able to handle the array given by my json data?

Thanks!

 - OS: Ubuntu 22.04.01
 - Thingsboard IoT Gateway version 3.1
 - Python version 3.10.4

samson0v, 2022-09-22:
Hi @wesmcpl, you can use your own custom converter. You can create it using this [instruction](https://thingsboard.io/docs/iot-gateway/custom/).

samson0v, 2022-09-29:
@wesmcpl, can we close the issue?

closed 2022-09-29 by wesmcpl