thingsboard/thingsboard-gateway #929: [HELP] Thingsboard Gateway correct RPC request over MQTT

peterandthejetsons, 2022-09-21:
Hello,

I have the Thingsboard Gateway v3.1 installed in a docker environment on a Ubuntu 20.04 server. I have some devices connected to Thingsboard Gateway throught central MQTT broker. I'd like to control these devices throught MQTT with RPC requests, but I'm always get the following error message:

``
""2022-09-21 15:09:43" - |ERROR| - [mqtt_connector.py] - mqtt_connector - server_side_rpc_handler - 743 - RPC not handled: {'device': 'shellyplus1pm-441793955bcc', 'data': {'id': 61, 'method': 'Switch.Set', 'params': False}}"
``

I can control the device with simple mosquitto client with the following command:

```
root@mosquitto:~# mosquitto_pub -h 192.168.1.13 -p 1884 -t 'shelly1pm/shellyplus1pm-441793955bcc/rpc' -m '{"method":"Switch.Set","params":{"id":0,"o
n":"false"}}' -u user -P password
```

My MQTT config is the following:

```
{
  "broker": {
    "name": "Default Local Broker",
    "host": "192.168.1.13",
    "port": 1884,
    "clientId": "ThingsBoard_gateway",
    "maxMessageNumberPerWorker": 10,
    "maxNumberOfWorkers": 100,
    "security": {
      "type": "basic",
      "username": "user",
      "password": "password"
    }
  },
    "mapping": [
    {
      "topicFilter": "shelly1pm/+/status/#",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=shelly1pm\/)(.*?)(?=\/status)",
        "deviceTypeTopicExpression": "Shelly_Plus_1PM",
        "timeout": 60000,
        "timeseries": [
          {
            "type": "string",
            "key": "output",
            "value": "${output}"
          },
          {
            "type": "string",
            "key": "id",
            "value": "${id}"
          },
          {
            "type": "string",
            "key": "apower",
            "value": "${apower}"
          },
          {
            "type": "string",
            "key": "voltage",
            "value": "${voltage}"
          },
          {
            "type": "string",
            "key": "current",
            "value": "${current}"
          },
          {
            "type": "string",
            "key": "aenergy",
            "value": "${*.total}"
          },
          {
            "type": "string",
            "key": "temperature",
            "value": "${*.tC}"
          },
          {
            "type": "string",
            "key": "by_minute",
            "value": "${*.by_minute}"
          }
        ]
      }
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": "Switch.Set",
      "reguestTopicExpression": "shelly1pm/${deviceName}/rpc'",
      "valueExpression": "{'method':'Switch.Set',params:{'id':0,'on':${params}}"
    }
  ]
}
```

I'd like to ask for your help in this issue. If you need further information, please let me know.

Thanks in advance.

samson0v, 2022-09-22:
Hi @peterandthejetsons, I saw some interesting detail)
[image]
Try to replace `g` with `q`.

peterandthejetsons, 2022-09-22:
Dear @samson0v!

Thank you, that was my fault, srry, its working like a charms now. Thank you once again.

Best regards,
Peter

closed 2022-09-22 by peterandthejetsons