thingsboard/thingsboard-gateway #932: [HELP] Telemetry data is only updated every 2 to 3 hours since some devices broke

redasdasd, 2022-09-23:
I have modbus configured in version 2.9. Everything was fine a few days ago. But several of the devices are broken and others are functioning normally. Updates of telemetry data became very slow. I set it to collect every 10 minutes. Telemetry data is only updated every 2 to 3 hours since some devices broke. The configuration of broken devices cannot be deleted in Modbus, as these devices will be repaired in the future. How should I get a working device back to 10 minute telemetry？I have to keep using version 2.9.
{
  "server": {
    "name": "Modbus1 Default Server",
    "type": "tcp",
    "host": "192.168.30.10",
    "port": 502,
    "timeout": 35,
    "methode": "socket",
    "byteOrder": "BIG",
    "retries": true,
    "retryOnEmpty": true,
    "retryOnInvalid": true,
    "devices": [
      {
        "unitId": 200,
        "deviceName": "Krackow_Smartlogger",
        "attributesPollPeriod": 86400000,
        "timeseriesPollPeriod": 600000,
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "attributes": [
          {
            "tag": "Serial Nummer",
            "type": "string",
            "functionCode": 3,
            "objectsCount": 10,
            "address": 40713
          }
       ],
      "timeseries": [
          {
            "tag": "P_AC",
            "type": "32int",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 40525,
            "divider": 1000
          },
          {
            "tag": "RPC_P_PERC",
            "type": "16uint",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 40428,
            "divider": 10
          },
          {
            "tag": "CO2",
            "type": "32uint",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 40523,
            "divider": 10000
          },
          {
            "tag": "Tenergy",
            "type": "32uint",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 40562,
            "divider": 10
          },
          {
            "tag": "Zenergy",
            "type": "32uint",
            "functionCode": 3,
            "objectsCount": 2,
[... 588 lines cut ...]
	}, {
		"tag": "Wechselrichter_Wirkleistung",
		"type": "32int",
		"functionCode": 3,
		"objectsCount": 2,
		"address": 32290,
		"divider": 1000
	}, {
		"tag": "Wechselrichter_Leistung",
		"type": "16uint",
		"functionCode": 3,
		"objectsCount": 1,
		"address": 32285,
		"divider": 100
	}, {
		"tag": "Wechselrichter_Status",
		"type": "16uint",
		"functionCode": 3,
		"objectsCount": 1,
		"address": 32287
	}]
}
    ]
  }
}

samson0v, 2022-09-27:
Hi @redasdasd, it seems that you are using the old version of Gateway. Please, update your gateway (version 3.1) via any accessible for your method.

samson0v, 2022-09-29:
@redasdasd, any updates?

samson0v, 2022-12-21:
@redasdasd, did you try?

closed 2023-03-01 by samson0v