thingsboard/thingsboard-gateway #934: [HELP] ThingsBoard as Modbus TCP server

serfreeman1337, 2022-09-24:
**Describe the issue**
I'm trying to expose thingsboard's device attributes vie modbus tcp server.
Modbus connector does have “slave” section, but it looks like it only exposes registers with constant 'value' field. 
Am I get it right that 'slave' option is only for constant values and I can't expose device attributes throught it ?

**Configuration**
<details>
  <summary>tb_gateway.yaml</summary>

```yaml
thingsboard:
  host: ***
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: ***
  qos: 1
storage:
 type: memory
 read_records_count: 100
 max_records_count: 100000
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: Modbus Test
    type: modbus
    configuration: modbus_test.json
```
</details>
<details>
  <summary>modbus_test.json</summary>

```json
{
  "master": {
    "slaves": []
  },
  "slave": {
    "type": "tcp",
    "host": "0.0.0.0",
    "port": 5026,
    "method": "socket",
    "deviceName": "test2-vfd-1",
    "deviceType": "delixi-vfd",
    "pollPeriod": 5000,
    "sendDataToThingsBoard": false,
    "byteOrder": "BIG",
    "wordOrder": "BIG",
    "unitId": 0,
    "values": {
      "holding_registers": [
        {
          "attributes":
          [
            {
              "tag": "b000",
              "type": "16int",
              "objectsCount": 1,
              "address": 45056,
              "value": 0
            },
            {
              "tag": "b001",
              "type": "16int",
              "objectsCount": 1,
              "address": 45057,
              "value": 0
            }
          ],
          "rpc": [
            {
              "tag": "setParam",
              "type": "16int",
              "functionCode": 6,
              "objectsCount": 1,
              "address": 40960,
              "value": 0
            }
          ]
        }
      ],
      "coils_initializer": [],
      "input_registers": [],
      "discrete_inputs": []
    }
  }
}
```
</details>

**Connector name:**
Modbus Connector

**Versions (please complete the following information):**
- OS: Arch Linux
- Thingsboard IoT Gateway version: 3.1
- Python version: 3.10.7

samson0v, 2022-09-27:
Hi @serfreeman1337, no, It isn't. The `slave` section raises up Modbus "server" that will create all the registers you configured and you can change the value of these registers using any other software and of course, these values will be sent to ThingsBoard.

samson0v, 2022-09-29:
@serfreeman1337, any updates? Can we close the issue?

serfreeman1337, 2022-10-01:
> and you can change the value of these registers using any other software and of course, these values will be sent to ThingsBoard.

I tried to change them by "Write Single Holding Register" function, but changes are not reflected back to thingsboard (didn't see any client attributes or shared attributes updated).

What I wanted to do is to expose thingsboard device's client attributes ("deviceName" parameter) as modbus tcp registers.  
"tag" - client attribute name of "deviceName" thingsboard device ?