thingsboard/thingsboard-gateway #935: [HELP] How to process whole data array in MQTT custom converter

devaskim, 2022-09-26:
**Describe the issue**
Data from my broker has the following format:
```
[false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false,false]
```
As I understand, by design MQTT connector [splits incoming data array](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/connectors/mqtt/mqtt_connector.py#L406) and pass each element separately. How about make this behaviour more configurable? For example, let converter to notify connector that the incoming data is needed without splitting or any other modification.

I suggested #927 to solve described case. What do you (@imbeacon and @samson0v) think about my implementation? Thanks

**Connector name (If you need help with some connector/converter):**
MQTT Connector

**Versions (please complete the following information):**
 - Thingsboard IoT Gateway version 3.1

samson0v, 2022-09-27:
Hi @devaskim, we plan to add this functionality to the default converter. So you can remake your PR so that the default converter can convert this type of data or wait until we will released this feature.  
Because what you have forces you to change the code to get the necessary value.

devaskim, 2022-09-27:
@samson0v I will re-implement by my own. But we need to discuss details: what data structure should default converter return in case of incoming data array? As of 3.1 it is expected that it should be single Python dict.

devaskim, 2022-09-27:
@samson0v Perhaps I found the answer by myself. Based on [these lines](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/gateway/tb_gateway_service.py#L661-L662) I think it would be ok if default converter will return array of Python dict-s. Do you agree?

devaskim, 2022-09-27:
@samson0v Done. Please, review

closed 2022-09-27 by samson0v