thingsboard/thingsboard-gateway #942: [BUG] tb-gateway parses the received message as “INF“.

zlwjzf, 2022-10-08:
**Describe the bug**
When I access an MQTT protocol device and receive a message such as "42646174611E010000", the tb-gateway will parse the message as "INF", resulting in my custom "MqttCustomConverter.py" cannot parse the message content accurately

zlwjzf, 2022-10-08:
I also encountered messages that were parsed into data of type float, but I forgot to save the exact message contents. Please help to provide solutions, thank you!

devaskim, 2022-10-11:
Is "42646174611E010000" string or number on the broker side? And is this data wrapped in JSON object, e.g. `{ "my_variable": "42646174611E010000" }`?

zlwjzf, 2022-10-11:
@devaskim Hi, thanks for your attention
1、"42646174611E010000" is a string on the broker side
2、This data is not included in json

devaskim, 2022-10-11:
As I understand [decoding algorithm](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/tb_utility/tb_utility.py#L31) for incoming mqtt data there is no quick fix for your case. 
MQTT Connector tries to interpret data as json-formatted and your string hits [this branch](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/tb_utility/tb_utility.py#L36) as number in string.

devaskim, 2022-10-11:
I think it makes sense to add new configuration parameter for MQTT connector called e.g. `processRawData` to pass data to converter without any encoding.

@imbeacon @samson0v What do you think about this?

samson0v, 2022-10-19:
Hi @zlwjzf, please, send your MQTT Connector config file.

closed 2022-10-19 by imbeacon