thingsboard/thingsboard-gateway #950: [BUG] Socket Connector can't connect two or more client devices.

qwyinxiang, 2022-10-14:
**Describe the bug**
I use socket connector as tcp server, and I want to connect 2 or more device. 
My socket.json is like this:
`{
    "name": "TCP Connector Example",
    "type": "TCP",
    "address": "192.168.200.129",
    "port": 50000,
    "bufferSize": 1024,
    "devices": [
      {
        "address": "192.168.32.58:50001",
        "deviceName": "QX-001",
        "deviceType": "airdevice",
        "encoding": "utf-8",
        "telemetry": [
          {
            "key": "temp",
            "byteFrom": 0,
            "byteTo": -1
          },
          {
            "key": "hum",
            "byteFrom": 0,
            "byteTo": 2
          }
        ]
      },
      {
        "address": "192.168.32.55:50002",
        "deviceName": "QX-002",
        "deviceType": "airdevice",
        "encoding": "utf-8",
        "telemetry": [
          {
            "key": "temp",
            "byteFrom": 0,
            "byteTo": -1
          },
          {
            "key": "hum",
            "byteFrom": 0,
            "byteTo": 2
          }
        ]
      },
    ]
  }`

First of all, I create an "airdevice" profile in device profile, and specified the rule chain.
[image]

Then, I started my gateway.
But, only one device has the device type "airdevice" and the telemetry data goes into my specified rule chain, another device's type is "default" and the telemtery data goes into the Root Rule Chain.
[image]

samson0v, 2022-10-14:
Hi @qwyinxiang. Can you, please, send the logs (replace all INFO to DEBUG level in logs.conf file)?

qwyinxiang, 2022-10-14:
@samson0v Sorry, I can't find any logs at thingsboard_gateway/logs/ in my docker.
[image]

I have just modify the the default devicetype from "default" to "airdevice"。But thingsboard received device is still "default" type. I suspect this is the problem of thingsboard.
And in the connected_devices.json, device type is "airdevice".

[image]

Next, I will tail after the telemtery thingsboard received, and confirm the device type thingsboard received.

samson0v, 2022-10-14:
@qwyinxiang if you are using docker, try to run the container without `-d` and take a screenshot of the logs.

samson0v, 2022-10-19:
@qwyinxiang any updates?

qwyinxiang, 2022-10-25:
I have resolved this problem, It is the problem of thingsboard, I restarted thingsboard JS executor microservice, and everything goes well。thank you very much.

closed 2022-10-25 by samson0v