thingsboard/thingsboard-gateway #952: [BUG] Modbus connector doesn't support multiple slaves corresponding to the same Thingsboard device

ferguscan, 2022-10-16:
**Describe the bug**
I want to have multiple Modbus slave entries for the same Thingsboard device (i.e. same device name).  This would, for example, allow me to poll different sets of Modbus registers at different rates.

When I try to do this, TB receives data from only one of the slaves.

**Connector name (If bug in the some connector):**
Modbus

**Error traceback (If available):**
None.  It fails silently.

**Versions (please complete the following information):**
TB Gateway official Docker image 3.0.1

samson0v, 2022-10-17:
Hi @ferguscan, send, please, your Modbus Connector config file.

samson0v, 2022-10-17:
@ferguscan, it is impossible to have the same ThingsBoard devices with default converter because device separation by unitId

closed 2022-10-19 by imbeacon