thingsboard/thingsboard-gateway #953: [HELP] How to implement ota on tb-gateway?

lincoln310, 2022-10-17:
**Describe the issue**
How to implement ota on tb-gateway? Do you have this play? What should i do if i want to add it?

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.4.2]
 - Python version[e.g. 3.9]

samson0v, 2022-10-17:
Hi @lincoln310, for OTA API you can use our [Python SDK](https://github.com/thingsboard/thingsboard-python-client-sdk).

lincoln310, 2022-10-18:
@samson0v Thanks! But what i want is : my device connect to tb-gateway, and get firmware from it. So the tb-gateway should work with ota message as other attribute message. Like a bridge.

samson0v, 2022-10-18:
@lincoln310, Gateway doesn't support OTA API for now.

lincoln310, 2022-10-18:
> @lincoln310, Gateway doesn't support OTA API for now.

Indeed, so what should i do to add it? Any suggestion? I can send a pull request when i finish.

imbeacon, 2022-10-18:
Hi @lincoln310,

Unforunately at the moment ThingsBoard doesn't have API to get firmware for devices through the gateway, we will discuss this point with team and let you know about the results.

lincoln310, 2022-10-19:
@imbeacon I need 2 function:
1  ota for my device. This should connect to tb or tb-edge
2  custom binary data process on custom topic from device. This should use tb-Gateway

Any suggestion?

imbeacon, 2022-10-19:
@lincoln310 ,

1. There is no API on ThenigsBoard for the gateway to get firmware for devices connected through it, so there is no way at the moment to implement this on the gateway by now, unfortunatly.
2. We are going to add different types for default mqtt converter like string, bytes in the nearest future, we will write here when add them.

lincoln310, 2022-10-21:
I can get latest attribute about firmware on device that connected to gw, when update firmware on web.
So maybe i can just impletment in tb-gateway like this:
subscribe topic v2/fw/request/${deviceName}/chunk/${chunkIdx}
and send http request to tb to get the response, 
and then send as mqtt msg to v2/fw/response/${deviceName}/chunk/${chunkIdx}
and then combind the chunk data.

How about it?

arthurgustin, 2024-03-07:
Any news on this topic ? I try to implement OTA with the gateway too

lincoln310, 2024-03-14:
> Any news on this topic ? I try to implement OTA with the gateway too

Sorry, i give it up.