thingsboard/thingsboard-gateway #959: [HELP]

Produmize, 2022-10-20:
We want if network connection between the gateway and thingsboard instance is lost that the gateway buffer the income data.
We have the standaard option configured "memory storage" with the standard values, but  when the connection is lost between the gateway and the thingsboard instance it's almost direct loosing data. I think because of the low buffering space what is configured. For me it's unclear what the settings mean, what does  "read_records_count" & "max_records_count" mean?

How can I see how much data is stored when the connection is lost?
Is there a way i can calculate the values I have to set for 6 hour of storage?

**Configuration** (Attach your configuration file)
storage:
  type: memory
  read_records_count: 10
  max_records_count: 1000

**Versions (please complete the following information):**
 - OS: PiOS
 - Thingsboard IoT Gateway version 2.7

samson0v, 2022-10-24:
Hi @Produmize, 
`read_records_count` - how many records Gateway will read simultaneously.
`max_records_count` - the maximum records number that will be stored in memory, so this parameter depends on how much RAM you have.

If you want to persist your data store you can use file store or SQLite (you can configure it using [documentation](https://thingsboard.io/docs/iot-gateway/configuration/#storage-configuration)).

samson0v, 2022-10-30:
@Produmize, We decided to create a CLI to manage the Gateway and actually execute the commands you requested.
This feature will have a lot of things to manage Gateway instance and devices.

Thanks for your interest in ThingsBoard IoT Gateway!

Produmize, 2022-10-31:
Is it possible when the storage is full and/or when the connection to the "cloud"
 is disturbed to stop odbc polling until the connection is reestablished?

samson0v, 2022-11-01:
Hi @Produmize, when the storage is full, the connector can't put new data into it as the exception is raised Storage is full.

Produmize, 2022-11-01:
> Hi @Produmize, when the storage is full, the connector can't put new data into it as the exception is raised Storage is full.

But the odbc connector keeps polling (we use persistent and the iterator data keeps running)  and the data is lost on the gateway

closed 2023-01-10 by imbeacon