thingsboard/thingsboard-gateway #961: [BUG]Modbus connector should be "registerCount" not "objectsCount"

qwyinxiang, 2022-10-26:
**Describe the bug**
First, My modbus.json is configured like below, I can only get value of address 0，can't get the value of address 1.
`    "timeseries": [
          {
            "tag": "humidity",
            "type": "16int",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 0
          },
          {
            "tag": "temperature",
            "type": "16int",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 1
          }
        ]
      }
    ]
  }
}`

Then, I replace “objectsCount” to "registerCount", I can get the value of  address 0 adn address 1。
My new modbus.json is configured like below:
`       "timeseries": [
          {
            "tag": "humidity",
            "type": "16int",
            "functionCode": 3,
            "registerCount": 1,
            "address": 0
          },
          {
            "tag": "temperature",
            "type": "16int",
            "functionCode": 3,
            "registerCount": 1,
            "address": 1
          }
        ]
      }
    ]`
The configure in the modbus_connector.py
[image]

samson0v, 2022-10-26:
Hi @qwyinxiang, there is no bug, argument `count` will get from `objectsCount` (see on the first get `OBJECTS_COUNT_PARAMETER`), then if there is no `objectsCount`, will get `registersCount`, if there is no, will get `registerCount`, if there is no, will get 1.

qwyinxiang, 2022-10-27:
ok, I know, thank you very much.

closed 2022-10-28 by samson0v