thingsboard/thingsboard-gateway #964: Wrong device type

yohnnylim, 2022-10-27:
Devices created by TB Gateway does not follow the device type configured in the connector files (devicePatternType)

I tried creating new devices using FTP connector and set it to a certain devicePatternType, but when the device is created it always under default profile.

 - OS: Windows 10 (Docker)
 - Thingsboard IoT Gateway version : 3.0.1
 - Python version: 3.9

samson0v, 2022-10-27:
Hi @yohnnylim, send, please, the FTP connector config file and logs.

samson0v, 2022-10-30:
@yohnnylim, any updates?

yohnnylim, 2022-10-31:
Hi @samson0v , sorry for the late reply, I did modify the codes in ftp_connector.py abit to read the whole content of the file and send to storage the whole content, below is my code in the ftp_connector.py

`def __process_paths(self, ftp):
        for path in self.paths:
            time_point = timer()
            if time_point - path.last_polled_time >= path.poll_period or path.last_polled_time == 0:
                configuration = path.config
                converter = FTPUplinkConverter(configuration)
                path.last_polled_time = time_point

                if '*' in path.path:
                    path.find_files(ftp)

                for file in path.files:
                    current_hash = file.get_current_hash(ftp)
                    if ((file.has_hash() and current_hash != file.hash)
                        or not file.has_hash()) and file.check_size_limit(ftp):
                        file.set_new_hash(current_hash)
                        handle_stream = io.BytesIO()
                        ftp.retrbinary('RETR ' + file.path_to_file, handle_stream.write)
                        handled_str = str(handle_stream.getvalue(), 'UTF-8')
                        handled_array = handled_str.split('\n')
                        convert_conf = {'file_ext': file.path_to_file.split('.')[-1]}
                        if convert_conf['file_ext'] == 'json':
                            json_data = simplejson.loads(handled_str)
                            if isinstance(json_data, list):
                                for obj in json_data:
                                    converted_data = converter.convert(convert_conf, obj)
                                    self.__gateway.send_to_storage(self.getName(), converted_data)
                                    self.statistics['MessagesSent'] = self.statistics['MessagesSent'] + 1
                                    log.debug("Data to GS Vault: %s", converted_data)
                            else:
                                converted_data = converter.convert(convert_conf, json_data)
                                self.__gateway.send_to_storage(self.getName(), converted_data)
                                self.statistics['MessagesSent'] = self.statistics['MessagesSent'] + 1
                                log.debug("Data to GS Vault: %s", converted_data)
                        else:
                            if path.txt_file_data_view == 'FILE':
                                filedata = {
                                    'deviceName':self.__config["paths"][0]["devicePatternName"],
                                    'deviceType':self.__config["paths"][0]["devicePatternType"],
                                    'attributes':[],
                                    'telemetry':[{
                                        'content': handled_str
                                    }]
                                }
                                self.__gateway.send_to_storage(self.getName(), file_data)
                                self.statistics['MessagesSent'] = self.statistics['MessagesSent'] + 1

                            else:  #TABLE or SLICED
                                for (index, line) in enumerate(handled_array):
                                    ftpfold = ftp.pwd()
                                    line = line.replace("\r", "")
                                    if line != '':
                                        if file.read_mode == File.ReadMode.PARTIAL and index >= cursor:
                                            converted_data = converter.convert(convert_conf, line)
                                            if index + 1 == len(handled_array):
                                                file.cursor = index
                                        else:
                                            converted_data = converter.convert(convert_conf, line)
                                        self.__gateway.send_to_storage(self.getName(), converted_data)
                                        self.statistics['MessagesSent'] = self.statistics['MessagesSent'] + 1

                        handle_stream.close()
                    
                    try:
                        log.info("File to delete: " + file.path_to_file)
                        ftp.delete(file.path_to_file)
                    except:
                        pass`

Basically I added some codes that says if the txt_file_data_view == 'FILE', the gateway will read the whole file and upload the whole content without going through uplink converter, the deviceType received in the rule chain when files are uploaded in this manner does not follow the devicePatternType configured in the ftp connector.

yohnnylim, 2022-11-01:
I think my issue is the same as this one here

https://github.com/thingsboard/thingsboard-gateway/issues/420

samson0v, 2022-11-01:
Hi @yohnnylim, we added functionality related to #420, you have to update your Gateway to the newest version 3.2 and add `handleDeviceRenaming: true` to tb_gateway.yaml config file.

Also, please send PR with your changes and we will review it.

yohnnylim, 2022-11-02:
I tried installing the new TB gateway 3.2, keeps flagging issues with tb_client.py timed out

""2022-11-02 04:24:32" - |ERROR| - [tb_client.py] - tb_client - run - 177 - timed out"
Traceback (most recent call last):
  File "/thingsboard_gateway/gateway/tb_client.py", line 172, in run
    self.client.connect(keepalive=keep_alive,
  File "/root/.local/lib/python3.10/site-packages/tb_device_mqtt.py", line 255, in connect
    self._client.connect(self.__host, self.__port, keepalive=keepalive)
  File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/local/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/usr/local/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
TimeoutError: timed out

yohnnylim, 2022-11-02:
installed and run TB Gateway 3.2, the problem is still there, the first time a device is created it follows the devicePatternType, then after removing the device, and re-provision another device with the same name, it will go back to default device profile.

yohnnylim, 2022-11-07:
Ok found the solution, need to use the latest TB version 3.4 also then everytime we delete the device an RPC will be sent to the gateway, 

I will submit my PR soon once I tested the codes.

Thanks

closed 2022-11-07 by yohnnylim