thingsboard/thingsboard-gateway #966: [HELP] the modbus connector doesn't work well

qwyinxiang, 2022-10-28:
My machine is ARMv7 architecture.
 I pulled the latest image and configured the modbus connector, but the docker failed to run properly。
The error logs are showing below:
`/thingsboard_gateway/gateway/tb_gateway_service.py", line 528, in _connect_with_connectors
    if connector_type.lower() != 'grpc' and 'Grpc' not in self._implemented_connectors[
AttributeError: 'NoneType' object has no attribute '__name__'
`

samson0v, 2022-10-30:
Hi @qwyinxiang, we are investigating this bug and let you know about the result.

samson0v, 2022-10-31:
Hi @qwyinxiang, try, please, the new Gateway version 3.2 and let us know about the result.

samson0v, 2022-12-05:
@qwyinxiang, any updates? Can we close the issue?

closed 2022-12-12 by samson0v