thingsboard/thingsboard-gateway #967: [BUG] KeyError: 'type' on custom MQTT converter

devaskim, 2022-10-28:
**Describe the bug**
MQTT connector fails to accept custom converter configuration if there is no `type` field with `json|raw` value
This bug appeared after #956

**Connector name (If bug in the some connector):**
MQTT converter

**Error traceback (If available):**
```
""2022-10-28 13:54:07" - |ERROR| - [mqtt_connector.py] - mqtt_connector - _on_connect - 364 - 'type'"
Traceback (most recent call last):
  File "C:\Users\User\code\thingsboard-gateway\thingsboard_gateway\connectors\mqtt\mqtt_connector.py", line 336, in _on_connect
    default_converters[mapping['converter']['type']])
KeyError: 'type'
```

samson0v, 2022-10-30:
Hi @devaskim, we will fix it in the nearest future (tomorrow)).

devaskim, 2022-10-31:
@samson0v @imbeacon 
I think PR #970 doesn't help because `mapping['converter']['type']` will be executed in any case, in other words, for all converter types.

Here is my converter configuration that cause exception
```
    {
      "topicFilter": "SuperMax/X3R04A80010/MeterMassConsumption/+",
      "converter": {
        "extension": "MeterConverter",
        "deviceNameTopicExpression": "(?<=SuperMax\/)(.*?)(?=\/MeterMassConsumption.*)",
        "deviceTypeTopicExpression": "Boiler"
      }
    }
```