thingsboard/thingsboard-gateway #968: Docker container won't start due to missing gcc

Eulenbaer, 2022-10-28:
**Describe the bug**
Today I pulled the recently published 3.2 docker image from docker-hub and andtried to do my first steps with thingsboard-iot-gateway.
The container fails to properly starts, menaing thingsboard-gateway does not come online.

Checking the docker logs I stumbled upon a pytrhon traceback indicating an error during 'mmh 3' module installation.
Apparently the module fails to install due to gcc missing. Please see the attached log.

**Error traceback**
```

× Encountered error while trying to install package.
      
╰─> mmh3

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
Collecting tb-mqtt-client
  Using cached tb_mqtt_client-1.4.1-py3-none-any.whl
Collecting mmh3
  Using cached mmh3-3.0.0.tar.gz (10 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: requests in /usr/local/lib/python3.10/site-packages (from tb-mqtt-client) (2.28.1)
Requirement already satisfied: paho-mqtt>=1.6 in /root/.local/lib/python3.10/site-packages (from tb-mqtt-client) (1.6.1)
Requirement already satisfied: simplejson in /usr/local/lib/python3.10/site-packages (from tb-mqtt-client) (3.17.6)
Requirement already satisfied: idna<4,>=2.5 in /usr/local/lib/python3.10/site-packages (from requests->tb-mqtt-client) (3.4)
Requirement already satisfied: charset-normalizer<3,>=2 in /usr/local/lib/python3.10/site-packages (from requests->tb-mqtt-client) (2.1.1)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.10/site-packages (from requests->tb-mqtt-client) (2022.9.24)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/local/lib/python3.10/site-packages (from requests->tb-mqtt-client) (1.26.12)
Building wheels for collected packages: mmh3
  Building wheel for mmh3 (setup.py): started
  Building wheel for mmh3 (setup.py): finished with status 'error'
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      running bdist_wheel
      running build
      running build_ext
      building 'mmh3' extension
      creating build
      creating build/temp.linux-x86_64-cpython-310
      gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/usr/local/include/python3.10 -c MurmurHash3.cpp -o build/temp.linux-x86_64-cpython-310/MurmurHash3.o
      error: command 'gcc' failed: No such file or directory
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mmh3
  Running setup.py clean for mmh3
Failed to build mmh3
Installing collected packages: mmh3, tb-mqtt-client
  Running setup.py install for mmh3: started
  Running setup.py install for mmh3: finished with status 'error'
  error: subprocess-exited-with-error
  
  × Running setup.py install for mmh3 did not run successfully.
  │ exit code: 1
  ╰─> [10 lines of output]
      running install
      /usr/local/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_ext
      building 'mmh3' extension
      creating build
      creating build/temp.linux-x86_64-cpython-310
      gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/usr/local/include/python3.10 -c MurmurHash3.cpp -o build/temp.linux-x86_64-cpython-310/MurmurHash3.o
      error: command 'gcc' failed: No such file or directory
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
      
╰─> mmh3

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
Traceback (most recent call last):
  File "/thingsboard_gateway/gateway/tb_client.py", line 22, in <module>
tb-mqtt-client library not found - installing...
    from tb_gateway_mqtt import TBGatewayMqttClient
ModuleNotFoundError: No module named 'tb_gateway_mqtt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_utility.py", line 145, in install_package
    result = check_call([executable, "-m", "pip", "install", package, "--upgrade", "--user"])
  File "/usr/local/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/bin/python', '-m', 'pip', 'install', 'tb-mqtt-client', '--upgrade', '--user']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "//./thingsboard_gateway/tb_gateway.py", line 17, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/thingsboard_gateway/gateway/tb_gateway_service.py", line 37, in <module>
    from thingsboard_gateway.gateway.tb_client import TBClient
  File "/thingsboard_gateway/gateway/tb_client.py", line 25, in <module>
    TBUtility.install_package('tb-mqtt-client')
  File "/thingsboard_gateway/tb_utility/tb_utility.py", line 147, in install_package
    result = check_call([executable, "-m", "pip", "install", package, "--upgrade"])
  File "/usr/local/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/bin/python', '-m', 'pip', 'install', 'tb-mqtt-client', '--upgrade']' returned non-zero exit status 1.

```

**Versions**
 - OS: Docker 20.10.3, build 55f0773 @ Synology DSM 7.1.1-42962 Update 2
 - Thingsboard IoT Gateway version: 3.2 (tag: latest, arch amd64)
 - Python version: 3.10.8

samson0v, 2022-10-30:
Hi @Eulenbaer, it seems that this issue started from and can be fixed from these sources:
https://github.com/pypa/pip/issues/8657
https://stackoverflow.com/questions/56905807/cant-install-mmh3-for-python3-7-1-on-win10x64-visual-c-build-tools-arent-r

closed 2022-10-31 by imbeacon

imbeacon, 2022-10-31:
Hi, @Eulenbaer 

New docker images pushed to the dockerhub, please try to pull them again and run. If issue appear please feel free to reopen this issue.