thingsboard/thingsboard-gateway #969: [HELP]

asao16, 2022-10-28:
Hello!
I have installed ThingsBoard on Window 10 and gateway with Docker. I use the OPCUA connector. I haven't been able to get the Gateway to connect to ThingsBoard and I don't know how to interpret the error, although the problem seems to be with the connector configuration, perhaps in the nodes.
I am new with these programs so I appreciate any help or recommendation you can give me.

**Configuration** (Attach your configuration file)
[e.g. MQTT Connector]
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/9891691/tb_gateway.txt)
[opcua.txt](https://github.com/thingsboard/thingsboard-gateway/files/9891694/opcua.txt)

**Error traceback (If it was raised):**
""2022-10-28 20:15:05" - |INFO| - [opcua_connector.py] - opcua_connector - run - 125 - OPC-UA connector OPC-UA Connector connected to server 192.168.1.101:53530/OPCUA/SimulationServer"

""2022-10-28 20:15:12" - |ERROR| - [opcua_uplink_converter.py] - opcua_uplink_converter - convert - 47 - 'path'"

Traceback (most recent call last):

  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_uplink_converter.py", line 35, in convert

    path = TBUtility.get_value(information["path"], get_tag=True)

KeyError: 'path'

""2022-10-28 20:15:14" - |ERROR| - [opcua_uplink_converter.py] - opcua_uplink_converter - convert - 47 - 'path'"

Traceback (most recent call last):

  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_uplink_converter.py", line 35, in convert

    path = TBUtility.get_value(information["path"], get_tag=True)

KeyError: 'path'

""2022-10-28 20:15:14" - |ERROR| - [opcua_connector.py] - opcua_connector - scan_nodes_from_config - 341 - 'key'"

Traceback (most recent call last):

  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 327, in scan_nodes_from_config

    self.__search_nodes_and_subscribe(device_info)

  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 351, in __search_nodes_and_subscribe

    information_key = information["key"]

KeyError: 'key'

""2022-10-28 20:15:15" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - argument of type 'NoneType' is not iterable"

""2022-10-28 20:15:15" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - argument of type 'NoneType' is not iterable"

""2022-10-28 20:23:30" - |ERROR| - [opcua_uplink_converter.py] - opcua_uplink_converter - convert - 47 - 'path'"

Traceback (most recent call last):

  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_uplink_converter.py", line 35, in convert

    path = TBUtility.get_value(information["path"], get_tag=True)

KeyError: 'path'

""2022-10-28 20:23:39" - |ERROR| - [opcua_uplink_converter.py] - opcua_uplink_converter - convert - 47 - 'path'"

Traceback (most recent call last):

  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_uplink_converter.py", line 35, in convert

    path = TBUtility.get_value(information["path"], get_tag=True)

KeyError: 'path'

""2022-10-28 20:23:39" - |ERROR| - [opcua_connector.py] - opcua_connector - scan_nodes_from_config - 341 - 'key'"

Traceback (most recent call last):

  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 327, in scan_nodes_from_config

    self.__search_nodes_and_subscribe(device_info)

  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-3.0.1-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 351, in __search_nodes_and_subscribe

    information_key = information["key"]

KeyError: 'key'

""2022-10-28 20:23:39" - |INFO| - [opcua_connector.py] - opcua_connector - run - 134 - Reconnected to the OPC-UA server - 192.168.1.101:53530/OPCUA/SimulationServer"

""2022-10-28 20:23:44" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - argument of type 'NoneType' is not iterable"

""2022-10-28 20:23:44" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - __send_to_storage - 573 - argument of type 'NoneType' is not iterable"

```

**Versions (please complete the following information):**
 - OS: [window 10]
 - Thingsboard IoT Gateway version [3.0.1]
 - Python version[3.9.9]
 - Docker version [20.10.14, build a224086]

Thanks.

samson0v, 2022-10-30:
Hi @asao16, we will investigate this bug and will you know about the result.

samson0v, 2022-11-01:
Hi @asao16, try to update the Gateway docker image to the newest version 3.2 and let us know about the result.

asao16, 2022-11-02:
It's done. I update to latest the gateway and docker. Now the error has changed to:

sock.connect(sa)
OSError: [Errno 99] Cannot assign requested address
""2022-11-02 19:04:11" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
""2022-11-02 19:04:11" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
""2022-11-02 19:04:11" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 1 for device *"
""2022-11-02 19:04:14" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import GrpcOpcUaConnector from /thingsboard_gateway/grpc_connectors/opcua."
""2022-11-02 19:04:14" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import OpcUaConnector from /thingsboard_gateway/connectors/opcua."
""2022-11-02 19:04:14" - |INFO| - [opcua_connector.py] - opcua_connector - open - 86 - Starting OPC-UA Connector"
""2022-11-02 19:04:14" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
""2022-11-02 19:04:14" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: ipconfig: not found
"
""2022-11-02 19:04:14" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: sw_vers: not found
"
""2022-11-02 19:04:14" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: uptime: not found
"
""2022-11-02 19:04:14" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: system_profiler: not found
"
""2022-11-02 19:04:14" - |ERROR| - [opcua_connector.py] - opcua_connector - __connect - 135 - Connection refused on connection to OPC-UA server with url 127.0.0.1:53530/OPCUA/SimulationServer"
""2022-11-02 19:04:17" - |WARNING| - [tb_updater.py] - tb_updater - check_for_new_version - 91 - Cannot connect to the update service. PLease check your internet connection."
""2022-11-02 19:04:18" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: ipconfig: not found
"
""2022-11-02 19:04:18" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: sw_vers: not found
"
""2022-11-02 19:04:18" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: uptime: not found
"
""2022-11-02 19:04:18" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: system_profiler: not found
"
""2022-11-02 19:04:21" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: ipconfig: not found
"
""2022-11-02 19:04:21" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: sw_vers: not found
"
""2022-11-02 19:04:22" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: uptime: not found
"
""2022-11-02 19:04:22" - |ERROR| - [statistics_service.py] - statistics_service - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: system_profiler: not found
"
""2022-11-02 19:04:24" - |ERROR| - [opcua_connector.py] - opcua_connector - __connect - 135 - Connection refused on connection to OPC-UA server with url 127.0.0.1:53530/OPCUA/SimulationServer"

samson0v, 2022-11-03:
@asao16, because you are using example statistics config file, so your OS hasn't this commands.
Simply delete `configuration: statistics.json` from `statistics` section in tb_gateway.yaml.

asao16, 2022-11-03:
It worked but now it seems to be stuck. I also don't think it was able to connect to ThingsBoard or even read the data from the OPC server.

""2022-11-03 16:08:17" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
""2022-11-03 16:08:17" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
""2022-11-03 16:08:17" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 1 for device *"
""2022-11-03 16:08:20" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import GrpcOpcUaConnector from /thingsboard_gateway/grpc_connectors/opcua."
""2022-11-03 16:08:20" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import OpcUaConnector from /thingsboard_gateway/connectors/opcua."
""2022-11-03 16:08:21" - |INFO| - [opcua_connector.py] - opcua_connector - open - 86 - Starting OPC-UA Connector"
""2022-11-03 16:08:21" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
""2022-11-03 16:08:24" - |WARNING| - [tb_updater.py] - tb_updater - check_for_new_version - 91 - Cannot connect to the update service. PLease check your internet connection."
""2022-11-03 16:08:33" - |INFO| - [opcua_connector.py] - opcua_connector - __connect - 133 - OPC-UA connector OPC-UA Connector connected to server 192.168.1.105:53530/OPCUA/SimulationServer"

samson0v, 2022-11-14:
@asao16 change all INFO to DEBUG in logs.conf file and send the logs.

asao16, 2022-11-14:
Hi @samson0v
It seems that it is able to read the nodes but cannot connect to the thingsboard.

The problem seems to be in this line:
|DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 206 - Disconnected client: <paho.mqtt.client.Client object at 0x7fa452d1a800>, user data: None, result code: 2"
""2022-11-14 17:59:09" - |DEBUG| - [tb_client.py] - tb_client - run - 170 - connecting to ThingsBoard"

The complete LOGS are:
""2022-11-14 17:58:19" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
""2022-11-14 17:58:19" - |DEBUG| - [tb_updater.py] - tb_updater - check_for_new_version - 75 - Checking for new version"
""2022-11-14 17:58:19" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
""2022-11-14 17:58:19" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 1 for device *"
""2022-11-14 17:58:19" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log service"
""2022-11-14 17:58:19" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log extension"
""2022-11-14 17:58:19" - |DEBUG| - [tb_client.py] - tb_client - run - 170 - connecting to ThingsBoard"
""2022-11-14 17:58:19" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log converter"
""2022-11-14 17:58:19" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log connector"
""2022-11-14 17:58:19" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log tb_connection"
""2022-11-14 17:58:19" - |DEBUG| - [memory_event_storage.py] - memory_event_storage - __init__ - 27 - Memory storage created with following configuration: 
Max size: 100000
 Read records per time: 100"
""2022-11-14 17:58:19" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_connector_keys - 1157 - Persistent keys file not found"
""2022-11-14 17:58:19" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 36 - Root path is: /thingsboard_gateway"
""2022-11-14 17:58:19" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 38 - Debian installation extensions folder exists."
""2022-11-14 17:58:19" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcOpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fa452d1b790>, origin='/thingsboard_gateway/connectors/opcua/opcua_converter.py')"
""2022-11-14 17:58:19" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcOpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fa452d1bd90>, origin='/thingsboard_gateway/connectors/opcua/opcua_connector.py')"
""2022-11-14 17:58:19" - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 206 - Disconnected client: <paho.mqtt.client.Client object at 0x7fa452d1a800>, user data: None, result code: 2"
""2022-11-14 17:58:20" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcOpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fa45390c250>, origin='/thingsboard_gateway/connectors/opcua/opcua_uplink_converter.py')"
""2022-11-14 17:58:20" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcOpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fa45390c1f0>, origin='/thingsboard_gateway/grpc_connectors/opcua/opcua_connector.py')"
""2022-11-14 17:58:20" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import GrpcOpcUaConnector from /thingsboard_gateway/grpc_connectors/opcua."
""2022-11-14 17:58:20" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='OpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fa45395c640>, origin='/thingsboard_gateway/connectors/opcua/opcua_converter.py')"
""2022-11-14 17:58:20" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='OpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fa45395c760>, origin='/thingsboard_gateway/connectors/opcua/opcua_connector.py')"
""2022-11-14 17:58:20" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import OpcUaConnector from /thingsboard_gateway/connectors/opcua."
""2022-11-14 17:58:20" - |INFO| - [opcua_connector.py] - opcua_connector - open - 86 - Starting OPC-UA Connector"
""2022-11-14 17:58:20" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_devices - 1199 - No device found in connected device file."
""2022-11-14 17:58:20" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __read_data_from_storage - 791 - Send data Thread has been started successfully."
""2022-11-14 17:58:20" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
""2022-11-14 17:58:20" - |DEBUG| - [tb_client.py] - tb_client - run - 170 - connecting to ThingsBoard"
""2022-11-14 17:58:21" - |DEBUG| - [tb_client.py] - tb_client - run - 170 - connecting to ThingsBoard"
""2022-11-14 17:58:21" - |DEBUG| - [opcua_connector.py] - opcua_connector - __connect - 124 - http://www.prosysopc.com/OPCUA/SampleAddressSpace"
""2022-11-14 17:58:21" - |DEBUG| - [opcua_connector.py] - opcua_connector - __connect - 125 - 6"
[... cut ...] itoredItemNotification(ClientHandle:{self.ClientHandle}, Value:{self.Value}))"
""2022-11-14 17:58:33" - |DEBUG| - [opcua_connector.py] - opcua_connector - datachange_notification - 675 - [SUBSCRIPTION] Data to ThingsBoard: {'deviceName': 'Temperatura de los gases de extracción', 'deviceType': 'default', 'attributes': [], 'telemetry': [{'ts': 1668448713000, 'values': {'temperatura': '800.0'}}]}"
""2022-11-14 17:58:33" - |DEBUG| - [tb_client.py] - tb_client - run - 170 - connecting to ThingsBoard"

samson0v, 2022-11-15:
@asao16 do you use ThingsBoard cloud or local installation?

samson0v, 2022-11-15:
@asao16 maybe it is some problem with your DNS or the IP address of your local ThingsBoard server is incorrect.

asao16, 2022-11-16:
It doesn't seem to be the case. I tried with different ip addresses, LAN, wireless, the result is the same except when I use localhost it returns an error. I also used those same addresses to access ThingsBoard from different computers and I was able to access without problems.

samson0v, 2022-11-22:
@asao16, from result code:
[image]

It seems that problem on your side.

closed 2022-12-21 by samson0v

andonoc, 2023-01-04:
> @asao16, because you are using example statistics config file, so your OS hasn't this commands. Simply delete `configuration: statistics.json` line from `statistics` section in tb_gateway.yaml.
> 
> Also, if you are using docker, dont use 127.0.0.1 in your config because it doesn't work.

Hi, i have the same errors

> ""2023-01-04 20:33:17" - |ERROR| - [statistics_service.py] - statistics_service                                                                              - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: uptime: not fo                                                                             und
"
""2023-01-04 20:33:17" - |ERROR| - [statistics_service.py] - statistics_service                                                                              - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: system_profile                                                                             r: not found
"
""2023-01-04 20:33:21" - |ERROR| - [statistics_service.py] - statistics_service                                                                              - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: ipconfig: not                                                                              found
"
""2023-01-04 20:33:21" - |ERROR| - [statistics_service.py] - statistics_service                                                                              - run - 67 - Statistic parameter raise the exception: /bin/sh: 1: sw_vers: not f  

Deleting  configuration: statistics.json line from statistics section removes the errors.
Does this hurt something?