thingsboard/thingsboard-gateway #975: [BUG]

riony, 2022-11-02:
**Description**
Latest release v3.2 Debian Installer using .zst compression and unable to install on raspbian 64-bit 

Installed on a clean installation of raspbian. I was able to install previous versions with no issues. Changing download flag from latest to 3.1 also worked on the same machine. 

**Traceback**
`$ sudo apt install ./python3-thingsboard-gateway.deb -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python3-thingsboard-gateway' instead of './python3-thingsboard-gateway.deb'
The following additional packages will be installed:
  figlet libabsl20200923 libc-ares2 libprotobuf23 python-pip-whl python3-cachetools python3-decorator python3-grpcio python3-jsonpath-rw python3-pip python3-ply python3-protobuf python3-pyfiglet python3-regex python3-simplejson
  python3-termcolor python3-wheel python3-yaml toilet-fonts
Suggested packages:
  python-ply-doc toilet
The following NEW packages will be installed:
  figlet libabsl20200923 libc-ares2 libprotobuf23 python-pip-whl python3-cachetools python3-decorator python3-grpcio python3-jsonpath-rw python3-pip python3-ply python3-protobuf python3-pyfiglet python3-regex python3-simplejson
  python3-termcolor python3-thingsboard-gateway python3-wheel python3-yaml toilet-fonts
0 upgraded, 20 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/6,680 kB of archives.
After this operation, 21.5 MB of additional disk space will be used.
Get:1 /home/adasiot/python3-thingsboard-gateway.deb python3-thingsboard-gateway all 3.2-1 [126 kB]
Selecting previously unselected package figlet.
(Reading database ... 41314 files and directories currently installed.)
Preparing to unpack .../00-figlet_2.2.5-3_arm64.deb ...
Unpacking figlet (2.2.5-3) ...
Selecting previously unselected package libabsl20200923:arm64.
Preparing to unpack .../01-libabsl20200923_0~20200923.3-2_arm64.deb ...
Unpacking libabsl20200923:arm64 (0~20200923.3-2) ...
Selecting previously unselected package libc-ares2:arm64.
Preparing to unpack .../02-libc-ares2_1.17.1-1+deb11u1_arm64.deb ...
Unpacking libc-ares2:arm64 (1.17.1-1+deb11u1) ...
Selecting previously unselected package libprotobuf23:arm64.
Preparing to unpack .../03-libprotobuf23_3.12.4-1_arm64.deb ...
Unpacking libprotobuf23:arm64 (3.12.4-1) ...
Selecting previously unselected package python-pip-whl.
Preparing to unpack .../04-python-pip-whl_20.3.4-4+rpt1+deb11u1_all.deb ...
Unpacking python-pip-whl (20.3.4-4+rpt1+deb11u1) ...
Selecting previously unselected package python3-cachetools.
Preparing to unpack .../05-python3-cachetools_4.2.1-1_all.deb ...
Unpacking python3-cachetools (4.2.1-1) ...
Selecting previously unselected package python3-decorator.
Preparing to unpack .../06-python3-decorator_4.4.2-2_all.deb ...
Unpacking python3-decorator (4.4.2-2) ...
Selecting previously unselected package python3-protobuf.
Preparing to unpack .../07-python3-protobuf_3.12.4-1_arm64.deb ...
Unpacking python3-protobuf (3.12.4-1) ...
Selecting previously unselected package python3-grpcio.
Preparing to unpack .../08-python3-grpcio_1.30.2-3_arm64.deb ...
Unpacking python3-grpcio (1.30.2-3) ...
Selecting previously unselected package python3-ply.
Preparing to unpack .../09-python3-ply_3.11-4_all.deb ...
Unpacking python3-ply (3.11-4) ...
Selecting previously unselected package python3-jsonpath-rw.
Preparing to unpack .../10-python3-jsonpath-rw_1.4.0-4_all.deb ...
Unpacking python3-jsonpath-rw (1.4.0-4) ...
Selecting previously unselected package python3-wheel.
Preparing to unpack .../11-python3-wheel_0.34.2-1_all.deb ...
Unpacking python3-wheel (0.34.2-1) ...
Selecting previously unselected package python3-pip.
Preparing to unpack .../12-python3-pip_20.3.4-4+rpt1+deb11u1_all.deb ...
Unpacking python3-pip (20.3.4-4+rpt1+deb11u1) ...
Selecting previously unselected package python3-pyfiglet.
Preparing to unpack .../13-python3-pyfiglet_0.8.0+dfsg-1_all.deb ...
Unpacking python3-pyfiglet (0.8.0+dfsg-1) ...
Selecting previously unselected package python3-regex.
Preparing to unpack .../14-python3-regex_0.1.20201113-1_arm64.deb ...
Unpacking python3-regex (0.1.20201113-1) ...
Selecting previously unselected package python3-simplejson.
Preparing to unpack .../15-python3-simplejson_3.17.2-1_arm64.deb ...
Unpacking python3-simplejson (3.17.2-1) ...
Selecting previously unselected package python3-termcolor.
Preparing to unpack .../16-python3-termcolor_1.1.0-3_all.deb ...
Unpacking python3-termcolor (1.1.0-3) ...
Selecting previously unselected package python3-yaml.
Preparing to unpack .../17-python3-yaml_5.3.1-5_arm64.deb ...
Unpacking python3-yaml (5.3.1-5) ...
dpkg-deb: error: archive '/tmp/apt-dpkg-install-WP6KIS/18-python3-thingsboard-gateway.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive /tmp/apt-dpkg-install-WP6KIS/18-python3-thingsboard-gateway.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
Selecting previously unselected package toilet-fonts.
Preparing to unpack .../19-toilet-fonts_0.3-1.3_all.deb ...
Unpacking toilet-fonts (0.3-1.3) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-WP6KIS/18-python3-thingsboard-gateway.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)`

**Versions:**
 - OS: Raspbian 64 bit Lite
 - Thingsboard IoT Gateway version V3.2

helsmore, 2022-11-02:
I've also encountered this bug, given the error message I suspect the package was produced on Ubuntu, which uses `zstd` compression by default; the version of `dpkg` included with Raspbian and Debian doesn't support this compression algorithm (per https://bugs.debian.org/892664).

The workaround:
```
sudo apt install zstd
wget https://github.com/thingsboard/thingsboard-gateway/releases/latest/download/python3-thingsboard-gateway.deb
mkdir deb-temp
cd deb-temp
ar x ../python3-thingsboard-gateway.deb
rm ../python3-thingsboard-gateway.deb
zstd -d *.zst
rm *.zst
xz *.tar
ar r ../python3-thingsboard-gateway.deb debian-binary control.tar.xz data.tar.xz
cd ..
rm -r deb-temp
sudo apt remove --purge zstd
sudo apt install ./python3-thingsboard-gateway.deb -y
```
These commands do the following:
1. Install the `zstd` CLI.
2. Download, unpack and delete the package file.
3. Decompress the files that were packaged used the `zstd` algorithm (results in `.tar` files).
4. Re-compress the `.tar` files using `xz`.
5. Re-create the package file, remove the temporary files.
6. Uninstall the `zstd` CLI.
7. Install the ThingsBoard IoT Gateway from the re-created package file.

imbeacon, 2022-11-02:
Deb package was updated. @helsmore thanks for workaround.

helsmore, 2022-11-02:
Thanks @imbeacon, I've compared the original and new deb packages for the v3.2 release:
[image]
The left-hand pane is the package for v3.2 you uploaded today, the right-hand pane is the original package for the v3.2 release (after `zstd` to `xz` conversion), as you can see there's some file differences and files newly added in the latest package, was the old package built from the wrong commit? Does this have any implications for the other packages produced for this release (RPM, Docker etc.)

riony, 2022-11-03:
Fix verified. Thanks for resolving issue.

imbeacon, 2022-11-03:
Hi @helsmore ,

I have built the new package, based on master branch, it already contains some additions, so this may case this difference.

closed 2022-11-03 by imbeacon