thingsboard/thingsboard-gateway #981: [Feature request] MQTT device filtering

devaskim, 2022-11-07:
**This ticket is to discuss the ways how such feature could be implemented.** And I am open to contribute. 

MQTT protocol support only `+` operator to grab all device topics. I would like to have more flexible way to filter devices.

I consider the following capabilities:
- RegExp and explicit name matching for white and black (i.e. filtered) device lists
- in-file (i.e mqtt.json) configuration set up
- remote configuration (RPC or shared attributes, I don't know yet)

In scope of implementation, I suggest to pass to converter some callback to check if device in white or black list.

@imbeacon @samson0v What do you think about this feature?

samson0v, 2022-11-14:
@devaskim, we decided to add new param `filterFile: list.json` it will be a file with two params: `whitelist` and `blacklist` where you can configure allowed and denied devices.
If you have other questions, feel free to ask.
Can you implement it and contribute?

devaskim, 2022-11-14:
Hi @samson0v 
For a past few days we re-built a bit our solution so at the moment device filtering feature is not so critical for us. Thus I could add this item to our roadmap but only with minor priority. 

Now it is vital for us to filter duplicates on gateway side. Here is [discussion](https://github.com/thingsboard/thingsboard-gateway/issues/986) (_please, reopen it_) I can help with implementation

ashvayka, 2022-12-30:
Please don't use colors in the list names. Use allowed/disallowed instead.

closed 2023-01-04 by imbeacon