thingsboard/thingsboard-gateway #984: [BUG] Gateway's version resolving need to be improved

devaskim, 2022-11-09:
**Describe the bug**
As of 3.2 version resolving is based on distribution information, but there is Docker deployment and manual/IDE running option where is no distribution.

**Error traceback (If available):**
```
[===UPDATE===]

 New version 3.1 is available!

[===UPDATE===]
"
""2022-11-09 10:49:48" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 84 - 3.1"
""2022-11-09 10:49:48" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 86 -

[===UPDATE===]

 New version 3.1 is available!

[===UPDATE===]

```

**Versions (please complete the following information):**
 - Thingsboard IoT Gateway version 3.2

devaskim, 2023-12-10:
@imbeacon The issue reproduces again. Seems like the following changes were reverted.
[image]

devaskim, 2024-01-30:
@imbeacon Please, reopen because of [reverting this pr](https://github.com/thingsboard/thingsboard-gateway/commit/65e756786adb209c7c818589a0c3ae67e6181347) it reproduces again.

imbeacon, 2024-01-30:
@devaskim, yes, it was reverted, because it causes an issue with remote configuration.

devaskim, 2024-01-30:
@imbeacon ok, but what about the original issue that I mentioned in the description?

devaskim, 2024-03-04:
@imbeacon Please, take a look at https://github.com/thingsboard/thingsboard-gateway/pull/1317