thingsboard/thingsboard-gateway #985: [HELP]Executing tb-gatewayV3.2, the following error is reported

qwyinxiang, 2022-11-10:
**Describe the issue**
When I executing tb-gateway v3.2. the following error is reported.
`"D:\Program Files\python\python.exe" E:/IOT/thingsboard/thingsboard-gateway-3.2/thingsboard_gateway/tb_gateway.py
""2022-11-10 11:31:18" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
""2022-11-10 11:31:18" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
""2022-11-10 11:31:18" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 1 for device *"
""2022-11-10 11:31:18" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import GrpcMqttConnector from E:\IOT\thingsboard\thingsboard-gateway-3.2\thingsboard_gateway\grpc_connectors\mqtt."
""2022-11-10 11:31:18" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import MqttConnector from E:\IOT\thingsboard\thingsboard-gateway-3.2\thingsboard_gateway\connectors\mqtt."
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted mapping handlers: 4"
Exception in thread Statistics Thread:
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected mapping handlers: 0"
Traceback (most recent call last):
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted serverSideRpc handlers: 2"
  File "D:\Program Files\python\lib\threading.py", line 973, in _bootstrap_inner
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected serverSideRpc handlers: 0"
    self.run()
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted connectRequests handlers: 2"
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected connectRequests handlers: 0"
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted disconnectRequests handlers: 2"
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected disconnectRequests handlers: 0"
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted attributeRequests handlers: 1"
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected attributeRequests handlers: 0"
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 243 - Number of accepted attributeUpdates handlers: 1"
""2022-11-10 11:31:18" - |INFO| - [mqtt_connector.py] - mqtt_connector - load_handlers - 247 - Number of rejected attributeUpdates handlers: 0"
""2022-11-10 11:31:18" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
  File "E:\IOT\thingsboard\thingsboard-gateway-3.2\thingsboard_gateway\gateway\statistics_service.py", line 63, in run
    process = subprocess.run(attribute['command'], stdout=subprocess.PIPE, stderr=subprocess.PIPE,
  File "D:\Program Files\python\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "D:\Program Files\python\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "D:\Program Files\python\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] 系统找不到指定的文件。
""2022-11-10 11:31:18" - |WARNING| - [tb_updater.py] - tb_updater - check_for_new_version - 91 - Cannot connect to the update service. PLease check your internet connection."
Exception in thread Thread-7:
Traceback (most recent call last):
  File "D:\Program Files\python\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "D:\Program Files\python\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\client.py", line 3591, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\client.py", line 1164, in _loop
    rc = self.loop_read()
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\client.py", line 3039, in _packet_handle
    return self._handle_connack()
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\client.py", line 3083, in _handle_connack
    reason = ReasonCodes(CONNACK >> 4, identifier=result)
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\reasoncodes.py", line 128, in __init__
    self.getName()  # check it's good
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\reasoncodes.py", line 174, in getName
    return self.__getName__(self.packetType, self.value)
  File "D:\Program Files\python\lib\site-packages\paho_mqtt-1.6.1-py3.9.egg\paho\mqtt\reasoncodes.py", line 138, in __getName__
    assert identifier in self.names.keys(), identifier
AssertionError: 5
""2022-11-10 11:31:19" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 2 for device *"
`
[image]

samson0v, 2022-11-15:
Hi @qwyinxiang, I think you have to delete the following line `configuration: statistics.json` in tb_gateway.yaml file.

qwyinxiang, 2022-11-17:
I had delete the line "configuration: statistics.json", and everything works well. thank you very much.@samson0v

closed 2022-11-17 by samson0v