thingsboard/thingsboard-gateway #986: [Feature request] MQTT connector: Provide option to send data only on change

devaskim, 2022-11-10:
**Describe the issue**
At least `Modbus` and `CAN` connectors already [has this options](https://thingsboard.io/docs/iot-gateway/config/modbus/) so there is no need to explain for what it is.

devaskim, 2022-11-14:
@imbeacon @samson0v Thanks for your time!
Please look at [my comment to PR](https://github.com/thingsboard/thingsboard-gateway/pull/991#issuecomment-1313446558)

devaskim, 2022-11-14:
@imbeacon @samson0v 
I forgot to mention it would be top of the top if such filtering module (shared between all connectors) would persist latest values to database in case of existing `storage` section is enabled. 
I know it is not the same but TB server already has separate database table for latest values. So gateway could go this way too.

devaskim, 2022-11-17:
@imbeacon @samson0v Please, take a look at #1000 (I am lucky - 1000th PR)

samson0v, 2022-11-22:
@devaskim see comments #1000