thingsboard/thingsboard-gateway #987: [BUG]The modbus connector will report an error after running for a while

qwyinxiang, 2022-11-11:
**Describe the bug**
The modbus connector will report an error after running for a while

**Connector name (If bug in the some connector):**
Modbus Connector

I use the modbus connector. My gateway is V3.1. When I started the gateway, modbus was able to get data, but after running it for a few hours, it was unable to get data, and the following error apears:

[image]

samson0v, 2022-11-21:
Hi @qwyinxiang, try to use the latest version of Gateway (3.2), there were bug fixes related to Modbus Connector.

samson0v, 2022-12-05:
@qwyinxiang any updates? Can we close the issue?

closed 2022-12-12 by samson0v