thingsboard/thingsboard-gateway #988: [HELP] odbc connection losing data

Produmize, 2022-11-11:
We did some testing on Thingsboard with the gateway software using odbc connector to see if there is any data-loss.

We setup a MS SQL database with 1 table with 4.500.000 records. The gateway was configured to poll the database every X seconds, collecting Y records a time. The first results showed 0,3% records missing in the Thingsboard environment (Postgres).

Additional testing polling 1 record a time showed even more missing data: z%

Do you recognize this behavior, can you advice on what we are doing wrong maybe?

configuration file:
[odbc.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/9991535/odbc.json.txt)

Connector name: ODBC Connector

Versions:
 - OS: PiOS (on RPi 4 4Gb)
 - Thingsboard IoT Gateway version 2.7-1
 - Python version 3.9

samson0v, 2022-12-05:
Hi @Produmize, we will investigate this situation.
Thanks for your interest in ThingsBoard IoT Gateway!

Produmize, 2022-12-05:
@samson0v 
Last week I was testing with it and it looks like incoming data is updating/overwriting it self.
Could it that when there is coming  to fast data thingsboard gives it the same timestamp if there is no time stamp defined?

It looks like I solved it by making a rulechain with a "transformation - script" node containing: metadata.ts = msg.TS;
Where TS = a uniq timestamp what comes with the data out of the database I'm connected to.
I only didn't find the time to test if it's solved with it.

In later versions I saw you could give the ts in the gateway with the data.

Niek

samson0v, 2023-09-29:
Hi @Produmize, this issue was fixed. Try the latest version from the master branch and let us know about the result.

closed 2024-06-19 by imbeacon