thingsboard/thingsboard-gateway #989: [HELP] How to deploy dockerized both TB server and mqtt gateway on same machine

devaskim, 2022-11-12:
**Describe the issue**
I understand that it is network port issue but don't know how to solve with minimal reconfiguration.
Gateway receives data from broker, but cannot connect to TB server due to `timed out`

**Configuration** (Attach your configuration file)
Defaults for TB server and gateway MQTT configuration (1883 port)

**Connector name:**
MQTT Connector

**Error traceback (If it was raised):**
```
TimeoutError: timed out
""2022-11-12 12:14:08" - |ERROR| - [tb_client.py] - tb_client - run - 177 - timed out"
Traceback (most recent call last):
  File "/thingsboard_gateway/gateway/tb_client.py", line 172, in run
    self.client.connect(keepalive=keep_alive,
  File "/root/.local/lib/python3.10/site-packages/tb_device_mqtt.py", line 255, in connect
    self._client.connect(self.__host, self.__port, keepalive=keepalive)
  File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/root/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/local/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/usr/local/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
```

**Versions:**
 - Thingsboard IoT Gateway version 3.2

samson0v, 2022-11-14:
Hi @devaskim you have to had two separate MQTT brokers (one for ThibgsBoard and another one for Gateway).

samson0v, 2022-11-22:
@devaskim, any updates?

devaskim, 2022-11-22:
Well, I tried suggested by me solution, but no luck. Maybe I was wrong in configuration. 
Now we are running gateway on our free machine, so I think I close this issue till next attemp)

closed 2022-11-22 by devaskim