thingsboard/thingsboard-gateway #990: [HELP] Socket Connector

mica83, 2022-11-14:
Hi,

i'm using Socket Connector : https://thingsboard.io/docs/iot-gateway/config/socket/

but after running ThingsBoard IoT gateway, i don't see the new device (for the socket) in my ThingsBoard instance.

The device should be appears automatically in the device list or not? Thanks

samson0v, 2022-11-15:
Hi @mica83, send, please, your logs (change all INFO to DEBUG in logs.conf file). Also, send your config files (tb_gateway.yaml and socket.json).

samson0v, 2022-11-15:
@SimoneG97, try to use `0.0.0.0` intead of `127.0.0.1` in socket.json.

SimoneG97, 2022-11-15:
Seems that did the trick guess docker was the culprit, thanks!
```
""2022-11-15 10:55:27" - |DEBUG| - [socket_connector.py] - socket_connector - run - 163 - New connection ('172.17.0.1', 33500) established"
""2022-11-15 10:55:27" - |DEBUG| - [socket_connector.py] - socket_connector - __process_tcp_connection - 185 - Connection ('172.17.0.1', 33500) closed"
```

A couple more questions:
1. Does the device appear automatically on the thingsboard dashboard or do I need to create it manually? (looks like this information is missing on the docs)
2. The previous log is showed after sending a "atr name" packet, but nothing returned and the connection closes shortly after. If I understood the docs correctly, with this command it should return me the value of the shared attribute "name". Is this correct?
If yes, how come nothing was returned instead of its value?
3. If I modify this shared attribute "name", will it send a TCP packet to the related device with this new value of "name"?
If no, is there a way to achieve this behavior?

@mica83 might add some questions that I forgot to ask.

samson0v, 2022-11-15:
@SimoneG97
1. Devices appear automatically only in the device list
    [image]
2.  Your device that is connected to the server can't be processed because of `not in config file` so you see AttributeError (maybe you did not correctly configure the IP address of a device in the config file).
3. Yes (notice that you have to have a handler for this message on your host).

SimoneG97, 2022-11-15:
Does the `not in config file` message refers to the socket.json or another file?
Because the device doesn't show in the device list, and when the device sends a packet, in the log it shows a connection with a different ip address (like shown previously):
```
""2022-11-15 11:26:58" - |DEBUG| - [socket_connector.py] - socket_connector - run - 163 - New connection ('172.17.0.1', 33620) established" <--- Here I was expecting 192.168.20.201 like in the socket.json, not 172.17.0.1
""2022-11-15 11:26:58" - |ERROR| - [socket_connector.py] - socket_connector - __process_data - 194 - Can't convert data from 172.17.0.1:33620 - not in config file"
```

samson0v, 2022-11-21:
Hi @SimoneG97, do you still need to make the port parameter optional?

SimoneG97, 2022-11-21:
I solved the problem by using a small API endpoint that will then communicate via TCP, I had to do it for other reasons related to the project in any case.
But for it to work correctly **in Docker**, at the very least the port shouldn't be searched for strictly via python dict .get() method.
Reasons:
1 - For Windows/Mac machines, Docker will change the IP address and port of the incoming packets, basically unusable sadly.
For example, if the packet comes from machine 192.4.50.4:3333, docker actually sees it as 172.17.0.X:\<different port\>.
This makes it fail the search in the configuration files [referring to .get(f"{address}:{port}")] since there you will put the actual address, not the "docker" adress.
2 - For Linux machines, IP address is preserved, but port is still changes in a seemingly random manner.
Like the example above, a packet from 192.4.50.4:3333 inside docker becomes 192.4.50.4:\<different port\>.
This also makes the .get() fail, but I tried to bypass it by making it optional.
Only then we have realized some problems related to our project and that it sadly needed an API endpoint in any case.
Just wanted to report this since it may help someone else trying to use the docker installation.

TL;DR: Docker does some really weird things with low level networking.

samson0v, 2022-11-22:
@SimoneG97, thanks for the information. We will add this warning to the documentation.

closed 2022-11-22 by samson0v