thingsboard/thingsboard-gateway #999: Trying to connect YABE simulator to BACnet thingsboard

Vaghulkumar95, 2022-11-17:
I'm trying to test out thingsboard-gateway and I'm stuck while trying to send data to thingsboard from Room Control Simulator from YABE. Is there something wrong with the address?

[image]

I got the Device IP and port from YABE. 

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**
pi@raspberrypi:~/thingsboard-gateway/thingsboard_gateway/config $ cat tb_gateway.yaml
thingsboard:
  host: thingsboard.cloud
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
    configuration: statistics.json
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: ********#gateway access token
   qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  #  -
  #   name: MQTT Broker Connector
  #   type: mqtt
  #   configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua_asyncio
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
  -
    name: BACnet Connector
    type: bacnet
    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Socket TCP/UDP Connector
#    type: socket
#    configuration: socket.json
#
#  -
#    name: XMPP Connector
#    type: xmpp
#    configuration: xmpp.json
#
#  -
#   name: OCPP Connector
#   type: ocpp
#   configuration: ocpp.json
#
# ========= Customization ==========
#
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
#
#  -
#    name: GRPC Connector 1
#    key: auto
#    type: grpc
#    configuration: grpc_connector_1.json

**Connector name (If you need help with some connector/converter):**
BACnet Connector

pi@raspberrypi:~/thingsboard-gateway/thingsboard_gateway/config $ cat bacnet.json
{
  "general": {
    "objectName": "TB_gateway",
    "address": "192.168.0.193:42837",#pi address and port, tried 0.0.0.0:47808 as well
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "BACnet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.0.110:54199",#from YABE
      "pollPeriod": 10000,
      "attributes": [
        {
          "key": "temperature",
          "type": "string",
          "objectId": "analogInput:0",
          "propertyId": "presentValue"
        },
    {
      "key": "temperature",
      "type": "string",
      "objectId": "analogInput:1",
      "propertyId": "presentValue"
    }
      ],
      "timeseries": [
        {
      "tag": "telemetry_1",    
          "key": "temperature",
          "type": "string",
          "objectId": "analogInput:0",
          "propertyId": "presentValue"
        },
    {
      "tag": "telemetry_2",
      "key": "temperature",
      "type": "string",
      "objectId": "analogInput:1",
      "propertyId": "presentValue"
    }
      ],
      "attributeUpdates": [
        {
          "key": "brightness",
          "requestType": "writeProperty",
          "objectId": "analogInput:1",
          "propertyId": "presentValue"
        }
      ],
      "serverSideRpc": [
        {
          "method": "set_state",
          "requestType": "writeProperty",
          "requestTimeout": 10000,
          "objectId": "analogInput:1",
          "propertyId": "presentValue"
        },
        {
          "method": "get_state",
          "requestType": "readProperty",
          "requestTimeout": 10000,
          "objectId": "analogInput:1",
          "propertyId": "presentValue"
        }
      ]
    }
  ]
}

**Error traceback (If it was raised):**
pi@raspberrypi:~/thingsboard-gateway/thingsboard_gateway/config $ thingsboard-gateway
Cannot load GRPC connector!
[STREAM ONLY] 2022-11-17 11:27:25,421 - ERROR - [mqtt_connector.py] - mqtt_connector - 285 - [Errno 111] Connection refused
[STREAM ONLY] 2022-11-17 11:27:25,588 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin/sh: 1: ipconfig: not found

[STREAM ONLY] 2022-11-17 11:27:25,608 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin/sh: 1: sw_vers: not found
[STREAM ONLY] 2022-11-17 11:27:25,662 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin/sh: 1: system_profiler: not found
connection FAIL with error 5 not authorised
[STREAM ONLY] 2022-11-17 11:27:29,432 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin/sh: 1: ipconfig: not found
[STREAM ONLY] 2022-11-17 11:27:29,451 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin/sh: 1: sw_vers: not found
[STREAM ONLY] 2022-11-17 11:27:29,498 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin/sh: 1: system_profiler: not found
[STREAM ONLY] 2022-11-17 11:27:33,262 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin/sh: 1: ipconfig: not found
[STREAM ONLY] 2022-11-17 11:27:33,277 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin/sh: 1: sw_vers: not found
[STREAM ONLY] 2022-11-17 11:27:33,321 - ERROR - [statistics_service.py] - statistics_service - 67 - Statistic parameter raise the exception: /bin/sh: 1: system_profiler: not found
[STREAM ONLY] 2022-11-17 11:27:35,434 - ERROR - [mqtt_connector.py] - mqtt_connector - 285 - [Errno 111] Connection refused

**Versions (please complete the following information):**
 - OS: Raspbian bullseye 11[e.g. Ubuntu 20.04] - installed thingsboard-gateway in raspberrypi 3b
 - Thingsboard IoT Gateway version 3.2 [e.g. 3.0.1]
 - Python version 3.10 [e.g. 3.9]

samson0v, 2022-11-21:
Hi @Vaghulkumar95, delete the following line from the tb_gateway.yaml file: `configuration: statistics.json`.

Vaghulkumar95, 2022-11-21:
Hi @samson0v , thanks for getting back, I was able to get thro' this by seeing few other issues related to bacnet(from here https://github.com/thingsboard/thingsboard-gateway/search?p=1&q=bacnet&type=issues) ,by changing the logs from INFO to DEBUG and in pi I was not in etc/thingsboard-gateway/config which is why I was not able to see any changes reflected. When I changed my path in raspberry pi and by changing back the IP to 0.0.0.0:47808 I was able to see the device created and the data flowing thro'.

I have few questions regarding the same.
1) so a new device got created for me, if I need to see the values for an existing device created in thingsboard how to do the same?
2) how to see historic data? and which api can I use to review the historic data and historical property of the asset? 

The same thing I wasn't able to achieve in docker. 
So in one post I saw that if the network doesn't match it might not be able to connect to the device. 
For eg. my network id is 192.168.0.110 and the docker is in 172.28.240.1 it wont be able to connect to the device right? 
So while running the docker command should I use a particular command so that it runs in the same IP? 

I tried this but it didn't work:

docker run -p 192.168.0.240:8080:80 -d -it -e host=thingsboard.cloud -e port=1883 -e accessToken=kLpgLXHLJgjixGzKBZjF  -v %HOMEPATH%/tb-gateway/config/thingsboard_gateway/config -v %HOMEPATH%/tb-gateway/extensions/thingsboard_gateway/extensions -v %HOMEPATH%/tb-gateway/logs/thingsboard_gateway/logs --name tb-gateway --restart always thingsboard/tb-gateway.

Vaghulkumar95, 2022-11-21:
[modbus logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/10056793/modbus.logs.txt)

I tried to stimulate Modbus slave & poll in PC and run Thingsboard in raspberry pi. Can you help me with this as well?

[image]

Vaghulkumar95, 2022-11-22:
Traceback (most recent call last):
  File "/root/.local/lib/python3.9/site-packages/twisted/internet/default.py", line 45, in _getInstallFunction
    from twisted.internet.pollreactor import install
  File "/root/.local/lib/python3.9/site-packages/twisted/internet/pollreactor.py", line 29, in <module>
    from twisted.internet import posixbase
  File "/root/.local/lib/python3.9/site-packages/twisted/internet/posixbase.py", line 16, in <module>
    from twisted.internet import error, tcp, udp
  File "/root/.local/lib/python3.9/site-packages/twisted/internet/tcp.py", line 99, in <module>
    from twisted.internet import abstract, address, base, error, fdesc, main
  File "/root/.local/lib/python3.9/site-packages/twisted/internet/base.py", line 34, in <module>
    from twisted.internet import abstract, defer, error, fdesc, main, threads
  File "/root/.local/lib/python3.9/site-packages/twisted/internet/defer.py", line 42, in <module>
    from typing_extensions import Literal, ParamSpec, Protocol
ImportError: cannot import name 'ParamSpec' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)

I had to install python3.10 to get this error out, and now I have the Modbus values passing.

Need your help in these :

1) A new device got created for me, if I need to see the values for an existing device created in thingsboard how to do the same?
2) how to see historic data? and which api can I use to review the historic data and historical property of the asset?

The same thing I wasn't able to achieve in docker.
So in one post I saw that if the network doesn't match it might not be able to connect to the device.
For eg. my network id is 192.168.0.110 and the docker is in 172.28.240.1 it wont be able to connect to the device right?
So while running the docker command should I use a particular command so that it runs in the same IP?

I tried this but it didn't work:

docker run -p 192.168.0.240:8080:80 -d -it -e host=thingsboard.cloud -e port=1883 -e accessToken=kLpgLXHLJgjixGzKBZjF -v %HOMEPATH%/tb-gateway/config/thingsboard_gateway/config -v %HOMEPATH%/tb-gateway/extensions/thingsboard_gateway/extensions -v %HOMEPATH%/tb-gateway/logs/thingsboard_gateway/logs --name tb-gateway --restart always thingsboard/tb-gateway.

And I'm trying to get the https://thingsboard.cloud/swagger-ui/#/telemetry-controller/getTimeseriesKeysUsingGET_1 I gave entityType as Attributes Key/Latest Telemetry Key and entityId as deviceID but I'm not sure if it is correct since I'm not getting any data back, can you let me know which values I should use for the same?

Vaghulkumar95, 2022-11-28:
Hi @samson0v, 
Any suggestions on the docker side of things?

Vaghulkumar95, 2022-11-29:
Hi @samson0v, 

I was able to run the docker in a specific IP address with the below command:

docker run -p 192.168.0.110:8082:80 -d -it --expose 8082 -e host=thingsboard.cloud -e port=1883 -e accessToken=kLpgLXHLJgjixGzKBZjF  -v %HOMEPATH%/tb-gateway/config/thingsboard_gateway/config -v %HOMEPATH%/tb-gateway/extensions/thingsboard_gateway/extensions -v %HOMEPATH%/tb-gateway/logs/thingsboard_gateway/logs --name tb-gateway --restart always thingsboard/tb-gateway

Now, the docker now has access to the LAN?

Is it possible to autostart thingsboard-gateway? Like a bat file for windows and an sh file for linux?

samson0v, 2022-12-01:
@Vaghulkumar95, if you want to autostart Gateway you have to use docker-compose with restart: always.

Vaghulkumar95, 2022-12-01:
Hi @samson0v, 

Thanks for getting back on this. And how to automate for linux?
And we are trying to stress test our gateway with thingsboard-iot-gateway to make sure our gateway is able to handle the load, do you have any suggestions for the same? 
For bacnet: 
I have loaded around 20 devices(YABE, BACNET room simulator) with 11 data points as of now and trying to get the CPUload, CPUTemperature with the python file.
[BacnetPython.txt](https://github.com/thingsboard/thingsboard-gateway/files/10130294/BacnetPython.txt)
[bacnet.txt](https://github.com/thingsboard/thingsboard-gateway/files/10130317/bacnet.txt)
Please find the json file above.

I need to stress test with Modbus as well.

Can you suggest me if there is something better I can do to stress test the gateway?

Vaghulkumar95, 2022-12-01:
[Bacnetlogslive.txt](https://github.com/thingsboard/thingsboard-gateway/files/10134130/Bacnetlogslive.txt)
Hi @samson0v ,

I have loaded around 30 devices(YABE, BACNET room simulator) with 11 data points. Unfortunately getting the below error:
""2022-12-01 16:17:25" - |ERROR| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - __init__ - 40 - [Errno 98] Address already in use"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 38, in __init__
    super().__init__(self.__device, self.__config["general"]["address"])
  File "/usr/local/lib/python3.9/dist-packages/bacpypes/app.py", line 535, in __init__
    self.mux = UDPMultiplexer(self.localAddress)
  File "/usr/local/lib/python3.9/dist-packages/bacpypes/bvllservice.py", line 96, in __init__
    self.directPort = UDPDirector(self.addrTuple)
  File "/usr/local/lib/python3.9/dist-packages/bacpypes/udp.py", line 155, in __init__
    self.bind(address)
  File "/usr/lib/python3.9/asyncore.py", line 326, in bind
    return self.socket.bind(addr)
OSError: [Errno 98] Address already in use

I have added the logs and the bacnet and tb_gateway.yaml in one file.

Currently testing a gateway if you could help me out on how to resolve the error that would be great !

samson0v, 2022-12-05:
@Vaghulkumar95, ensure that you correctly close Gateway and haven't running up one.

Vaghulkumar95, 2022-12-05:
@samson0v so I do,
systemctl stop thingsboard-gateway
systemctl restart thingsboard-gateway
sudo thingsboard-gateway

is there something else I have to do to close the gateway?

samson0v, 2022-12-07:
@Vaghulkumar95 try to manually find a port in use (in your case 47808):
`sudo netstat -tnlp | grep :47808`
and kill it:
`sudo kill <pid_of_process>`

Vaghulkumar95, 2022-12-09:
Hi @samson0v,
Thanks for that. If I need to install thingsboard-iot-gateway in windows machine, if I have Ubuntu inside that can I do this(https://thingsboard.io/docs/iot-gateway/install/deb-installation/)? 
I did install the same in ubuntu and I got the below error while running sudo thingsboard-gateway
[error.txt](https://github.com/thingsboard/thingsboard-gateway/files/10194236/error.txt)

Any thoughts on the error?

Why I am asking this is I'm not able to install WSL/WSL2 because the Operating System is Windows 10 Enterprise LTSC 64-bit (10.0, Build 17763) (17763.rs5_release.180914-1434). We have many Windows gateways that are currently at client sites and upgrading each to them to 18362 will be quite challenging.

samson0v, 2022-12-12:
@Vaghulkumar95 from your error logs, these may help:
https://github.com/pypa/pip/issues/8657
https://stackoverflow.com/questions/56905807/cant-install-mmh3-for-python3-7-1-on-win10x64-visual-c-build-tools-arent-r

Vaghulkumar95, 2022-12-18:
Hi @samson0v,

Yes thankyou, I was able to resolve that.
I'm trying this now with a new raspberrypi. Can you help me on why this isn't working?

[raspberrypi1 logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/10253515/raspberrypi1.logs.txt)

samson0v, 2022-12-21:
@Vaghulkumar95, did you expose 47808 port in docker file (or docker-compose file)?

Vaghulkumar95, 2022-12-21:
@samson0v, Sorry, I didn't get you. The logs which I sent is for raspberrypi and everything on this was done in pi.

samson0v, 2022-12-21:
@Vaghulkumar95, can we close the issue?

Vaghulkumar95, 2022-12-21:
Do I need to create a new one?

samson0v, 2022-12-21:
@Vaghulkumar95, no.
I can't understand what the problem is. Gateway can't connect to ThingsBoard?

Vaghulkumar95, 2022-12-21:
pi@raspberrypi:/etc/thingsboard-gateway/config $ sudo thingsboard-gateway
Cannot load GRPC connector!
""2022-12-18 09:23:56" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
""2022-12-18 09:23:56" - |DEBUG| - [tb_updater.py] - tb_updater - check_for_new_version - 75 - Checking for new version"
""2022-12-18 09:23:56" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
""2022-12-18 09:23:56" - |DEBUG| - [tb_client.py] - tb_client - run - 170 - connecting to ThingsBoard"
""2022-12-18 09:23:56" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 1 for device *"
""2022-12-18 09:23:56" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log service"
""2022-12-18 09:23:56" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log extension"
""2022-12-18 09:23:56" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log converter"
""2022-12-18 09:23:56" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log connector"
""2022-12-18 09:23:56" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log tb_connection"
""2022-12-18 09:23:56" - |INFO| - [event_storage_reader.py] - event_storage_reader - read_state_file - 148 - FileStorage_reader -- Initializing from state file: [./data/data_1671354725906.txt:0]"
""2022-12-18 09:23:56" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_connector_keys - 1157 - Persistent keys file not found"
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 36 - Root path is: /usr/local/lib/python3.10/site-packages/thingsboard_gateway"
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 38 - Debian installation extensions folder exists."
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcBACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb5eeb508>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_converter.py')"
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcBACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb5eeb208>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"
""2022-12-18 09:23:56" - |DEBUG| - [tb_client.py] - tb_client - _on_connect - 128 - TB client <paho.mqtt.client.Client object at 0xb536cbc8> connected to ThingsBoard"
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcBACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb2ca9b38>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_downlink_converter.py')"
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcBACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb2ca9b20>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb2ca9d60>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_converter.py')"
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb2ca9ce8>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"
""2022-12-18 09:23:56" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import BACnetConnector from /usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet."
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='BACnetUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb2c516a0>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_converter.py')"
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='BACnetUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb2c51688>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"
""2022-12-18 09:23:56" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 69 - Service subscription to topic v1/gateway/attributes - successfully completed."
""2022-12-18 09:23:56" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='BACnetUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb536ce68>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_downlink_converter.py')"
""2022-12-18 09:23:56" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 69 - Service subscription to topic v1/gateway/attributes/response - successfully completed."
[... cut ...]
""2022-12-18 09:23:56" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __read_data_from_storage - 791 - Send data Thread has been started successfully."
""2022-12-18 09:23:56" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
""2022-12-18 09:23:56" - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1671354725906.txt!"
""2022-12-18 09:23:57" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attributes_parse - 381 - Received data: {}"

After this there is no data flowing thro'.

Vaghulkumar95, 2022-12-23:
Hi @samson0v, Any reason on why the values are not flowing through?

samson0v, 2022-12-23:
Hi @Vaghulkumar95 found a bug, fixing...

Vaghulkumar95, 2022-12-23:
Understood, this is a go-live device so any updates on when this would be fixed or any alternate advice is highly appreciated.

Vaghulkumar95, 2022-12-23:
I got this error parallelly |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 223 - connection FAIL with error 5 not authorised" and found this https://github.com/thingsboard/thingsboard-gateway/issues/862. But I'm unaware on where to add this property ACTORS_MAX_CONCURRENT_SESSION_PER_DEVICE, is it on tb_gateway.yaml or in bacnet.json? Do you have an sample example where this property is used?

Vaghulkumar95, 2022-12-27:
Hi @samson0v,

Any suggestions on this? and anything on the bug?

Vaghulkumar95, 2023-01-02:
Hi @samson0v,

Anything suggestions on this error "connection FAIL with error 5 not authorised" and anything on the bug?

samson0v, 2023-01-02:
Hi @Vaghulkumar95, `"connection FAIL with error 5 not authorised"` - it seems you are using the wrong Gateway token. Also, make sure that Gateway uses the correct config files (not default).

Vaghulkumar95, 2023-01-02:
Hi @samson0v, 

Sure let me check on that once and anything on the bug which you mentioned earlier?

Vaghulkumar95, 2023-01-03:
Hi @samson0v,

I tried again today still NOT receiving the values. Pls find the below logs:

pi@raspberrypi:/etc/thingsboard-gateway/config $ sudo thingsboard-gateway
Cannot load GRPC connector!
""2023-01-03 10:23:58" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
""2023-01-03 10:23:58" - |DEBUG| - [tb_updater.py] - tb_updater - check_for_new_version - 75 - Checking for new version"
""2023-01-03 10:23:58" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
""2023-01-03 10:23:58" - |DEBUG| - [tb_client.py] - tb_client - run - 170 - connecting to ThingsBoard"
""2023-01-03 10:23:58" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 185 - Subscribed to *|* with id 1 for device *"
""2023-01-03 10:23:58" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log service"
""2023-01-03 10:23:58" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log extension"
""2023-01-03 10:23:58" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log converter"
""2023-01-03 10:23:58" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log connector"
""2023-01-03 10:23:58" - |DEBUG| - [tb_logger.py] - tb_logger - __init__ - 39 - Added remote handler to log tb_connection"
""2023-01-03 10:23:58" - |INFO| - [event_storage_reader.py] - event_storage_reader - read_state_file - 148 - FileStorage_reader -- Initializing from state file: [./data/data_1672306721092.txt:0]"
""2023-01-03 10:23:58" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __load_persistent_connector_keys - 1157 - Persistent keys file not found"
""2023-01-03 10:23:58" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 36 - Root path is: /usr/local/lib/python3.10/site-packages/thingsboard_gateway"
""2023-01-03 10:23:58" - |DEBUG| - [tb_loader.py] - tb_loader - find_paths - 38 - Debian installation extensions folder exists."
""2023-01-03 10:23:58" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcBACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb531f6d0>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_converter.py')"
""2023-01-03 10:23:58" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcBACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb4a16448>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"
""2023-01-03 10:23:58" - |DEBUG| - [tb_client.py] - tb_client - _on_connect - 128 - TB client <paho.mqtt.client.Client object at 0xb531f0e8> connected to ThingsBoard"
""2023-01-03 10:23:59" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 69 - Service subscription to topic v1/gateway/attributes - successfully completed."
""2023-01-03 10:23:59" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 69 - Service subscription to topic v1/gateway/attributes/response - successfully completed."
""2023-01-03 10:23:59" - |DEBUG| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - _on_subscribe - 69 - Service subscription to topic v1/gateway/rpc - successfully completed."
""2023-01-03 10:23:59" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcBACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb22bf610>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_downlink_converter.py')"
""2023-01-03 10:23:59" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='GrpcBACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb22bf5f8>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_uplink_converter.py')"
""2023-01-03 10:23:59" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb22bf838>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_converter.py')"
""2023-01-03 10:23:59" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='BACnetConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb22bf7f0>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"
""2023-01-03 10:23:59" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import BACnetConnector from /usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet."
""2023-01-03 10:23:59" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='BACnetUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb22bffb8>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_converter.py')"
""2023-01-03 10:23:59" - |DEBUG| - [tb_loader.py] - tb_loader - import_module - 59 - ModuleSpec(name='BACnetUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb22e2040>, origin='/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_connector.py')"
[... cut ...]
""2023-01-03 10:23:59" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - __read_data_from_storage - 791 - Send data Thread has been started successfully."
""2023-01-03 10:23:59" - |INFO| - [event_storage_reader.py] - event_storage_reader - delete_read_file - 172 - FileStorage_reader -- Cleanup old data file: ./data/data_1672306721092.txt!"
""2023-01-03 10:23:59" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
""2023-01-03 10:23:59" - |DEBUG| - [tb_gateway_service.py] - tb_gateway_service - _attributes_parse - 381 - Received data: {}"

Vaghulkumar95, 2023-01-03:
@samson0v tried this in one more pi with debian 10 and same result. Sill NOT receiving data.
[image]

Vaghulkumar95, 2023-01-05:
Hi @samson0v,

Can you also help me on how to (service daemons) autostart thingsboard gateway in debian during reboot?

samson0v, 2023-01-06:
Hi @Vaghulkumar95, 
As a rule, device data usually collect to timeseries (see official documentation below):
Attributes - https://thingsboard.io/docs/user-guide/attributes/
Timeseries - https://thingsboard.io/docs/user-guide/telemetry/

> Can you also help me on how to (service daemons) autostart thingsboard gateway in debian/linux during reboot?(the commands)

For example: https://www.digitalocean.com/community/tutorials/how-to-configure-a-linux-service-to-start-automatically-after-a-crash-or-reboot-part-1-practical-examples

Or you can google it for more information.

Vaghulkumar95, 2023-01-13:
Hi @samson0v,

Thank you so much for that. Can you help me on how to send Server RPC requests from thingsboard cloud to the device?

samson0v, 2023-01-13:
Hi @Vaghulkumar95, read the official documentation about Gateway RPCs and device RPCs:
https://thingsboard.io/docs/iot-gateway/guides/how-to-use-gateway-rpc-methods/
https://thingsboard.io/docs/iot-gateway/guides/how-to-use-get-set-rpc-methods/

Vaghulkumar95, 2023-01-16:
Hi @samson0v ,

Below is my modbus.json file.

pi@raspberrypi:~ $ vi /etc/thingsboard-gateway/config/modbus.json
    "wordOrder": "BIG",
    "unitId": 1,
    "attributes": [
            {
              "address": 0,
              "type": "16int",
              "tag": "Attribute1",
              "functionCode": 3,
              "objectsCount": 1
            }
          ],
          "timeseries": [
            {
              "address": 0,
              "type": "16int",
              "tag": "Telemetry1",
              "objectsCount": 1,
              "functionCode": 3
            },
            {
              "address": 1,
              "type": "16int",
              "tag": "Telemetry2",
              "objectsCount": 1,
              "functionCode": 3
            }
          ],
          "attributeUpdates": [
          {
            "tag": "maxTemperature",
            "type": "16int",
            "functionCode":3,
            "objectsCount": 1,
            "address": 0
          }
          ],
           "rpc": [
            {
            "tag": "maxTemperature_read",
            "type": "16int",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 0
          },
          {
            "tag": "setValue",
            "type": "16int",
            "functionCode": 6,
            "objectsCount": 1,
            "address": 1
          }
         ]
    }
  ]
}
}

In the RPC debug I'm getting the below error. Is there something wrong in the way I'm doing?
Raspberry-pi-3> maxTemperature_read
{"error":"maxTemperature - connector not found in available connectors.","code":404}

samson0v, 2023-01-17:
Hi @Vaghulkumar95, for example:
We have the device `Temp Sensor` that has configured RPC `maxTemperature_read`.
We have to configure `EntityAlies` to this device, as in the screenshot below:
[image]
Saving it and in the result our RPC debug terminal has a following look:
[image]
So from now, we can use our configured RPCs.

Vaghulkumar95, 2023-01-17:
Yes I was able to get that working, setvalue is working too, but I want to know how to implement it using rest api's as well. 
I tried for getvalue it is returning data, but not sure for setvalue it isn't working. How should the json body be , for setvalue?
[image]

samson0v, 2023-01-17:
@Vaghulkumar95, all these settings you can find in the [official documentation](https://thingsboard.io/docs/user-guide/rpc/).

And before asking a question, try to find the answer in docs or google it.

samson0v, 2023-01-26:
@Vaghulkumar95, can we close the issue or you have some other questions?

Vaghulkumar95, 2023-01-26:
Hi @samson0v,

I was able to get the rest api working. In params I gave the value directly and it started working. {"method":"setValue","params":78.78,"persistent":false,"timeout":5000}.

Yes, I was trying to connect two Advantech devices one has Ubuntu installed and the other one has Windows 10. So I'm trying to get Modbus RTU values from windows 10 device having a modbus simulator and the other one is receiving it thro COM ports.

{
  "master":
  {
    "slaves":
    [
  {
    "name": "COMPort1",
    "type": "serial",
    "method": "rtu",
    "port": "/dev/tty0",
    "baudrate": 115200,
    "pollPeriod": 5000,
    "wordOrder": "BIG",
    "byteOrder": "BIG",
    "timeout": 35,
    "unitId": 1,
    "deviceName": "Device1_Com1_Port",
    "attributesPollPeriod": 5000,
    "timeseriesPollPeriod": 5000,
    "sendDataOnlyOnChange": false,   
    "timeseries":
    [
        {
             "tag": "Telemetry1",
             "type": "16int",
             "functionCode": 3,
             "objectsCount": 1,
             "address": 0
        },
        {
             "tag": "Telemetry2",
             "type": "16int",
             "functionCode": 3,
             "objectsCount": 1,
             "address": 1
        },
       {
             "tag": "Telemetry3",
             "type": "16int",
             "functionCode": 3,
             "objectsCount": 1,
             "address": 2
        }

    ]
  }
 ]
}
}
advantech8869@KSA4358869:/etc/thingsboard-gateway/config$ sudo cat tb_gateway.yaml
thingsboard:
  host: thingsboard.cloud
  port: 1883
  remoteShell: true
  remoteConfiguration: false
  statistics:
    enable: false 
    statsSendPeriodInSeconds: 3600
    configuration: statistics.json
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: 6dwPtkAPC8D3WMzzAteC
  qos: 1
storage:
#  type: memory
#  read_records_count: 100
#  max_records_count: 100000
  type: file
  data_folder_path: ./data/
  max_file_count: 10
  max_read_records_count: 10
  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
# -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json
#  -
#   name: Modbus Connector
#   type: modbus
#   configuration: modbus.json
  -
    name: Modbus Connector
    type: modbus
    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua_asyncio
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Socket TCP/UDP Connector
#    type: socket
#    configuration: socket.json
#
#  -
#    name: XMPP Connector
#    type: xmpp
#    configuration: xmpp.json
#
#  -
#   name: OCPP Connector
#   type: ocpp
#   configuration: ocpp.json
#
# ========= Customization ==========
#
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
#
#  -
#    name: GRPC Connector 1
#    key: auto
#    type: grpc
#    configuration: grpc_connector_1.json

[image]

Am I giving the "port": "/dev/tty0", wrong?

[image]

samson0v, 2023-02-07:
Hi @Vaghulkumar95, very bad quality in the last screenshot, I can't read the error

Vaghulkumar95, 2023-02-07:
[image]

samson0v, 2023-02-07:
@Vaghulkumar95, you have some problems with your configuration
[image]

Vaghulkumar95, 2023-02-07:
Which configuration are you referring to?
[image]
[image]
[image]

samson0v, 2023-02-07:
@Vaghulkumar95, it seems that your Gateway using the wrong config files:
[image]

Vaghulkumar95, 2023-02-13:
[modbus_serial(sample).txt](https://github.com/thingsboard/thingsboard-gateway/files/10722261/modbus_serial.sample.txt)

Hi @samson0v,

I tried in the above format and it started working.
Currently I have a device called Advantech - EPC-U2117, I have 2 LAN cable ports.
One I connected(192.168.8.113) to one router and trying to read the modbus_serial meter data connected to a meter with RS485 to USB connector.
I have another Lan connected to my laptop and from here I configured it to a different network 169.254.130.3 and in the Advantech device I have configured it to 169.254.130.4 and in my laptop I have a modbus simulator trying to simulate the data.
In my modbus.json I have the IP address to 169.254.130.3 with port 502 and in my simulator I have the IP address as 169.254.130.4 with port 502. If I ping the address 169.254.130.4 I'm receiving packets, but I'm unable to connect my simulator and the device. Can you help me with this?

Vaghulkumar95, 2023-02-15:
Hi @samson0v,

Any suggestions on the above scenario?
And we ran into a similar issue to https://github.com/thingsboard/thingsboard-gateway/issues/844
[image]
where we have several devices are connected to one IP, and we want to monitor each device whether it is offline or online.
I see that the bacpypes library can't discover that variant of device connectivity, but is there any other thoughts on how to overcome this scenario?

samson0v, 2023-02-16:
@Vaghulkumar95, using the same IP address but different ports - is one single solution.

Vaghulkumar95, 2023-02-16:
Can bacpypes take network number + device ID?
When I enter 370080:94.10.1.1 in, it just shows no data, no error, just no response.
Is it network number + IP or network number + device ID?
Can you let us know on how this works?
[image]

samson0v, 2023-02-20:
@Vaghulkumar95, no, it can't.

closed 2023-03-01 by samson0v

Vaghulkumar95, 2023-03-01:
Thank you so much @samson0v for your help!

kamisettychandan, 2023-10-19:
@Vaghulkumar95 @samson0v 

Statement: I want to send Room Control Simulator data to Things Board Platform

Successful steps:
------------------
1. TB-Gateway created in ThingsBoard Platform
2. TB-Gateway installed in Laptop using Docker & Connected (using Access Token)
3. Device (for receiving Simulator data) configured in ThingsBoard Platform
4. Device connected to TB-Gateway (updated "provisionDeviceKey & provisionDeviceSecret" in tb-gateway.json)
5. From Device Connectivity (on HTTP), observing dummy data
6. Observed Simulator data in YABE

Unsuccessful steps:
---------------------
1. Simulator data not reached to ThingsBoard Platform
2. Observing "Rule Chain Failure". how to resolve issue. May be its affected

Assumptions:
--------------
1. Shall we update Device Name & Attributes in Device configuration as per YABE?

Please find Snaps below:

Snap 1: Simulator data in YABE
[image]

Snap 2: Device & Gateway Connection
[image]

Snap 3: Device Connectivity
[image]

Snap 4: Rule Chain Failure
[image]

Configuration Files:
--------------------
[bacnet.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/13037772/bacnet.json.txt)
[tb_gateway.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/13037773/tb_gateway.json.txt)
[list.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/13037776/list.json.txt)

I am not understanding, where I did mistake. Kindly refer guide me...

kamisettychandan, 2023-11-06:
@samson0v 
can you provide your suggestions

kamisettychandan, 2024-03-07:
@Vaghulkumar95,

Kindly help me...!
As per Docker Logs, BACnet connector not detecting 'Room Control Simulator' device. Could you help me...

Please find logs:
[Day 4 Docker_Logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/14523215/Day.4.Docker_Logs.txt)

Please find config logs:
[bacnet.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/14523193/bacnet.json.txt)
[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/14523194/tb-gateway.txt)