thingsboard/thingsboard-gateway #1006: [HELP] Example of using Timestamp (ts) in MQTT.json

sambartle, 2022-11-21:
[MQTT Connector]

Please can someone include an example or explain how to use the timestamp (ts) in timeseries data passed via the MQTT.json converter in the gateway? 
I have tried to add ts to a few places in the example MQTT.json config (and then pass it up, but it does not seem to be respected)

**Configuration** (Attach your configuration file)
Using the standatd MQTT.json expmaple supplied with thingsbaord-gateway and the following mapping:

  "mapping": [
    {
      "topicFilter": "topic/+/data",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=topic\/)(.*?)(?=\/data)",
        "deviceTypeTopicExpression": "et301",
        "timeout": 60000,
        "timeseries": [
          {
            "type": "double",
            "key": "key1",
            "value": "${p1}"
          },
          {
            "type": "double",
            "key": "key2",
            "value": "${p2}"
          }
        ]
      }
    }
  ]

Passing the MQTT message:
{"ts":1669035610000, "values":{"p1":4.5,"p2":45.0}}

If i remove the timestamp the data works as expected. (Example simplified)

**Versions (please complete the following information):**
 - OS: Debian
 - Thingsboard IoT Gateway version: Latest

sambartle, 2022-11-21:
Thank you @samson0v , I have tried this and it doesn't seem to work.. 

If i send: {"p1": 4.5, "p2": 45.0} the data is successfully logged. once I add the ts value, the data is simply ignored. 

Do i need to 'map' to the ts value as a timestamp in the config perhaps somehow?

sambartle, 2022-11-21:
After doing some digging I get the following int he gateway logs:

Without the timestamp:
2022-11-21 16:21:52,045 - DEBUG - [mqtt_connector.py] - mqtt_connector - 352 - Successfully converted message from topic topic/CD_inside/data

With the timestamp:
2022-11-21 16:22:20,507 - DEBUG - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 79 - Regular expression result is None. deviceTypeTopicExpression will be interpreted as a deviceType Topic: topic/CD_inside/data Regex: et301

samson0v, 2022-11-22:
@sambartle, as you can see from the screenshot, data is not ignored:
[image]

[image]

sambartle, 2022-11-22:
Apologies @samson0v, ignored was a bad choice of word. I meant that once I add the timestamp, thingsboard never logs the data, however with the timestamp removed it works fine. 

I have been experimenting overnight and I have found that it works as expected with shorter messages, but not longer ones.. 
Perhaps I should include the whole config as I simplified it in my initial post.. 

There are 27 parameters passed in my JSON MQTT message to the gateway, and this is the config to load those 27 parameters. If I send just the 27 parameters, they pass through the gateway and get sent to thingsboard against the correct device as intended. 

When adding the timestamp it will not work if the Connector config has 27 datapoints, but it DOES work if it only has 14/15 datapoints.. (and is otherwise identical... )

Here is the complete 27 item MQTT Connector configuration for the message below:

<pre>
    {
      "topicFilter": "et/+/datats",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=et\/)(.*?)(?=\/datats)",
        "deviceTypeTopicExpression": "(.*?)(?=\/.*\/datats)",
        "timeout": 60000,
        "timeseries": [
          {
            "type": "double",
            "key": "MeasPercentSmoothed",
            "value": "${p1}"
          },
          {
            "type": "double",
            "key": "MeasMgNm3",
            "value": "${p2}"
          },
          {
            "type": "double",
            "key": "MeasMgM3",
            "value": "${p3}"
          },
          {
            "type": "double",
            "key": "ETSCFm",
            "value": "${p4}"
          },
          {
            "type": "double",
            "key": "ETSCFc",
            "value": "${p5}"
          },
          {
            "type": "double",
            "key": "ETSPresSmpl",
            "value": "${p6}"
          },
          {
            "type": "double",
            "key": "ETSTempSmpl",
            "value": "${p7}"
          },
          {
            "type": "double",
            "key": "InSituSPANCalPIDConSetPoint",
            "value": "${p8}"
          },
          {
            "type": "int",
            "key": "SIG_PRE_AMP_HW_GAIN",
            "value": "${p9}"
          },
          {
            "type": "int",
            "key": "SIG_POST_AMP_HW_GAIN",
            "value": "${p10}"
          },
[... 162 lines cut ...]
          {
            "type": "boolean",
            "key": "msPowerUp",
            "value": "${p11}"
          },
          {
            "type": "boolean",
            "key": "msZeroCal",
            "value": "${p12}"
          },
          {
            "type": "boolean",
            "key": "msSpanCal",
            "value": "${p13}"
          },
          {
            "type": "boolean",
            "key": "msInSituSpanCal",
            "value": "${p14}"
          }
      }
    }
</pre>

Both messages work including the timestamp as expected.

samson0v, 2022-11-22:
@sambartle, ok, will investigate

john2151, 2023-07-05:
Similar issue here with TB-Gw 3.3 and UTC timestamps
The strange thing for me is, that it also breaks mappings for other topics. As soon as i change the topic filter for the MQTT-Stream with timestamps to something without traffic, the other topic comes back to life.
Also noteworthy is, that the data passes through to Thingsboard if it is configured under attributes instead of timeseries.

Any ideas?

Makodan, 2024-05-05:
I faced with the similar issue. Thingsboard doesn't process telemetry with timestamps. If I remove the ts from the telemetry message, it works. Any news on this?

imbeacon, 2024-12-11:
Hi,

The issue relates to old version of the gateway, problems like this are not appear on new version, please use it instead. Due to ts - ThingsBoard can handle only unix timestamp in milliseconds as ts, other formats are not supported at the moment.

closed 2024-12-11 by imbeacon