thingsboard/thingsboard-gateway #1012: [Feature] Enable/disable gateway data repack

devaskim, 2022-12-01:
**Describe the issue**
Gateway repacks all outgoing data based on `maxPayloadSizeBytes` value. I suggest to provide `enable/disable` option to use or not this algorithm at all and use `disable` by default.

One of our cases is array of alarm objects that gateway receives from MQTT broker. We need to deliver it to server as it is, i.e. without splitting some alarm objects.

I understand that I could increase `maxPayloadSizeBytes` value but from performance and reliability perspectives it would be better to just not to use repack algorithm at all.

samson0v, 2022-12-01:
Hi @devaskim, it is not necessary because we will migrate to protobuf and the message format will probably be changed.

devaskim, 2022-12-01:
Maybе... but:
1. How soon it happens?
2. Are you sure that later won't be `maxPayloadSizeBytes `?

samson0v, 2022-12-01:
@devaskim, also if the data repack function is off, ThingsBoard will be banned message that is more than 1024 bytes.

devaskim, 2022-12-01:
Thanks @samson0v I didn't know about that. I hope it is configurable parameter in server configuration, isn't it?

Also you convinced me that repack algorithm shouldn't be disabled by default

samson0v, 2022-12-01:
Yes, it is configurable but only for local ThingsBoard

samson0v, 2022-12-01:
@devaskim can we close the issue?

devaskim, 2022-12-02:
> Yes, it is configurable but only for local ThingsBoard

@samson0v In server's `thingsboard.yml` I fount [only this size option](https://github.com/thingsboard/thingsboard/blob/master/application/src/main/resources/thingsboard.yml#L713) in mqtt section: 
`max_payload_size: "${NETTY_MAX_PAYLOAD_SIZE:65536}"`

samson0v, 2022-12-05:
@devaskim I think that you have to configure this parameter on your MQTT Broker.

devaskim, 2022-12-05:
> I think that you have to configure this parameter on your MQTT Broker.

MQTT broker configuration doesn't relate to our discussion.

> also if the data repack function is off, ThingsBoard will be banned message that is more than 1024 bytes.

As I understood your prev-prev answer gateway's repack algorithm is only to deal with server side MQTT transport limitation. But after digging into server's configuration and code, I found max payload size is set to 65 536 B.

samson0v, 2022-12-05:
@devaskim my bad, in the default configuration, it is set to 65536 but for the TB cloud is 1024.

devaskim, 2022-12-05:
@samson0v Ok, I trust you)

closed 2022-12-05 by devaskim