thingsboard/thingsboard-gateway #1013: [BUG]

KRuok, 2022-12-02:
**Describe the bug**
Hi there, i have been testing my customized mqtt converter which returns a list of dict_result like what you possibly get from JsonMqttUplinkConverter. 
[image]

However, a list result cannot be accepted and failed the check  in MqttConnector.run() as list object has no get method

**Connector name (If bug in the some connector):**
[MQTT Connector]

**Error traceback:**
```
Traceback (most recent call last):
  File "/opt/conda/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/opt/conda/lib/python3.7/site-packages/thingsboard_gateway-3.2-py3.7.egg/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 857, in run
    if converted_data and (converted_data.get(ATTRIBUTES_PARAMETER) or
AttributeError: 'list' object has no attribute 'get'
```

**Versions:**
 - OS: [Ubuntu 18.04]
 - Thingsboard IoT Gateway version [3.2]
 - Python version[e.g. 3.7]

samson0v, 2022-12-02:
Hi @KRuok, converter can't process the list, you have to return dict object. See official [doc](https://thingsboard.io/docs/reference/mqtt-api/#telemetry-upload-api).

KRuok, 2022-12-05:
thanks @samson0v, no more questions for now

closed 2022-12-05 by KRuok