thingsboard/thingsboard-gateway #1014: [HELP] Device connectivity status

devaskim, 2022-12-02:
**Disclaimer**
I know that it is [server side feature](https://thingsboard.io/docs/user-guide/device-connectivity-status/), but that team doesn't like chatting with community so I decided to start discussion here.

**Describe the issue**
The main problem is when IoT platform (based on TB) provides alarm notifications (e.g. emails). When your gateway has been stopped for some reason, customer and/or support team would receive dozens of email about theirs device is offline despite only gateway has wrong behaviour.

Could it be improved in some way? For example, trigger *Inactivity event* for each device only if gateway is considered as online.
Or at least provide gateway option on server side to switch between suggested above approach and that is already implemented.

imbeacon, 2022-12-02:
Hi @devaskim,

Thanks for your message, we will discuss this point internally and decide how to solve this issue.
But actually at the moment you can try to solve it using rule chains, with logic like:

receive inactivity event -> Save device name and type to metadata -> check relation "Created" for device (In the future we are going to add relation "Connected" between device and gateway, because of "Created" relation creates by first gateway that device connected through) -> If it exists -> Change originator to related, using duplicate to related rule node -> get server attribute "active" of the gateway -> If it is true -> change originator to the device back, using device name and type from the metadata and rule node change originator -> create alarm

At the moment - logic described above is a workaround for your issue in some case (Except case if device can connect to different gateways)

devaskim, 2022-12-02:
Thanks @imbeacon I am considering the same approach but don't test yet. In scope of described approach I guess there might be race condition between inactivity of gateway and device event with the same `inactivityTimeout` (but it is only assumption)

imbeacon, 2022-12-02:
Yes, it can appear, but it just means that you should have inactivityTimeout for the gateway less than for device.

devaskim, 2023-03-21:
> Thanks for your message, we will discuss this point internally and decide how to solve this issue.

Hi @samson0v 
If no any other suggestions/improvements, let's close the ticket

closed 2023-03-21 by samson0v