thingsboard/thingsboard-gateway #1015: [HELP] Minimum Hardware Specs that can run TB IoT Gateway ? Open for Outsourcing

232485, 2022-12-03:
**Issue**
I would like to run the Thingsboard IoT Gateway in BLE configuration. Since SBCs such as RPIs is unavailable in supply, I am looking for alternates.

Can a strip-down version of IoT Gateway Stack (BLE) run on an ESP32 footprint or any microcontroller? If not, then what is the recommended alternate, such as Onion Omega?

Features required:

1. Microcontroller or hardware that can run Thingsboard IoT Gateway 
2. Scanning of BLE beacons- upto 50-100 at a time
3. Sending data via Cellular Network via 5G/LTE
4. Purpose: Industrial use.

**Connector name**
BLE

samson0v, 2022-12-05:
Hi @232485, you can run ThingsBoard IoT Gateway on the raspberry pi or other (Orange PI, etc.).
We plan to support Gateway for the ESP platform but it is not yet clear when the development starts.

samson0v, 2022-12-07:
@232485 any additional questions?

closed 2022-12-21 by samson0v