thingsboard/thingsboard-gateway #1016: [HELP] How to get Temperature and Humidity from SHT20 sensor

11208, 2022-12-03:
**Describe the issue**
Create description about your issue, and your actions to solve it.
I have succcesfully connect modbus device -- SHT20(XY-MD02) to tb_gateway. But the thing is only temperature data reach TB server. 

**Configuration** (Attach your configuration file)
tb_gateway.yaml
```
thingsboard:
  host: IP
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
#    configuration: statistics.json
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: TOKEN
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
"/etc/thingsboard-gateway/config/tb_gateway.yaml" 137L, 2659B                                                                                                                       16,33         Top
```
modbus.json
```
{
  "master": {
    "slaves": [
      {
        "unitId": 1,
        "deviceName": "Station-1",
        "deviceType": "sensors",
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "timeseries": [
          {
            "tag": "temperature",
            "type": "16uint",
            "functionCode": 4,
            "registerCount": 2,
            "address": 1,
            "devider": 10
          },
          {
            "tag": "humidity",
            "type": "16uint",
            "functionCode": 4,
            "registerCount": 2,
            "address": 2,
            "divider": 10
          }
        ],
        "host": null,
        "port": "/dev/ttyS0",
        "type": "serial",
        "method": "rtu",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "baudrate": 9600,
        "pollPeriod": 5000
      }
    ],
    "slave": null
  }
}

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]
MODBUS Converter

**Error traceback (If it was raised):**
Log from converter.log
```
""2022-12-03 21:01:14" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 84 - Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)"
NoneType: None
```

**Versions (please complete the following information):**
 - OS: RPI 3B+ Debian
 - Thingsboard IoT Gateway version 3.2
 - Python version 3.9

ahmed-0011, 2022-12-03:
16uint register type has  the data stored in one register
32uint register type has  data stored in two registers
check sensor manual either you need to change  registerCount to 1 or type to 32uint/32int.
But I think its 16uint with registerCount =1

11208, 2022-12-04:
Thanks for your reply.
After play around the data type and change registerCount like you suggest, setting like below will only output "temperature"
```
"timeseries": [
          {
            "tag": "temperature",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 1,
            "divider": 10
          },
          {
            "tag": "humidity",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 2,
            "divider": 10
          }
        ],
```

But when I run each address on different time, it return the correct value.

This return correct temperature data
```
"timeseries": [
          {
            "tag": "temperature",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 1,
            "divider": 10
          }
        ],
```
```
""2022-12-04 07:40:31" - |INFO| - [modbus_connector.py] - modbus_connector - run - 651 - {'deviceName': 'Station-1', 'deviceType': 'sensors', 'telemetry': [{'temperature': 30.0}], 'attributes': []}"
```

This return correct humidity data
```
"timeseries": [
          {
            "tag": "humidity",
            "type": "16int",
            "functionCode": 4,
            "registerCount": 1,
            "address": 2,
            "divider": 10
          }
        ],
```
```
""2022-12-04 07:38:14" - |INFO| - [modbus_connector.py] - modbus_connector - run - 651 - {'deviceName': 'Station-1', 'deviceType': 'sensors', 'telemetry': [{'humidity': 64.6}], 'attributes': []}"```

When both in the same timeseries array, humidity not working.

11208, 2022-12-04:
Thanks. You can close this.

closed 2022-12-04 by samson0v