thingsboard/thingsboard-gateway #1021: [BUG] Unexpected behaviour when writing on variables with variables with OPC-UA

SimoneG97, 2022-12-07:
1. When writing an a mapped variable under a variable, the device doesn't get updated unless the deviceNodePattern points directly at that variable containing a variable
2. In the attributes_updates section, attributeOnDevice doesn't work if defined with namespace + node identifier, only with node path.
Is this normal?

**Connector name (If bug in the some connector):**
OPC-UA Connector

**Error traceback (If available):**
No traceback available, even with all loggers set to DEBUG

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version: 2 weeks old from this day (I had not found an easy way to determine the gateway version, using docker)
 - Python version: 3.8.10

Configuration example (using Prosys):
```
{
	"server": {
		"name": "OPC-UA Prosys Server",
		"url": "192.2.90.2:53530",
		"timeoutInMillis": 5000,
		"scanPeriodInMillis": 5000,
		"disableSubscriptions": true,
		"subCheckPeriodInMillis": 1000,
		"showMap": true,
		"security": "None",
		"identity": {
			"type": "anonymous"
		},
		"mapping": [
			{
				"deviceNodePattern": "${ns=3;s=85/0:Simulation}",
				"deviceNamePattern": "Prosys",
				"attributes": [],
				"timeseries": [
					{
						"key": "Sawtooth",
						"path": "${Root\\.Objects\\.Simulation\\.Sawtooth}"
					}
				],
				"rpc_methods": [],
				"attributes_updates": [
					{
						"attributeOnThingsBoard": "StrTest",
						"attributeOnDevice": "Root\\.Objects\\.Simulation\\.StringTest"
					},
					{
						"attributeOnThingsBoard": "StrTest2",
						"attributeOnDevice": "Root\\.Objects\\.Simulation\\.TOBJ\\.Param"
					},
					{
						"attributeOnThingsBoard": "V1",
						"attributeOnDevice": "Root\\.Objects\\.Simulation\\.TOBJ\\.TOBJ2\\.TOBJ3\\.V1"
					},
					{
						"attributeOnThingsBoard": "V2",
						"attributeOnDevice": "Root\\.Objects\\.Simulation\\.TOBJ\\.TOBJ2\\.F1\\.V2"
					},
					{
						"attributeOnThingsBoard": "SubVar",
						"attributeOnDevice": "Root\\.Objects\\.Simulation\\.TOBJ\\.Param\\.SubVar"
					},
					{
						"attributeOnThingsBoard": "V4",
						"attributeOnDevice": "Root\\.Objects\\.Simulation\\.SimulationChild\\.SB1\\.V4"
					}
				]
			},
			{
				"deviceNodePattern": "${ns=3;i=1009}",
				"deviceNamePattern": "Prosys var with vars",
				"attributes": [],
				"timeseries": [
					{
						"key": "TS",
						"path": "${ns=3;i=1019}"
					}
				],
				"rpc_methods": [],
				"attributes_updates": [
					{
						"attributeOnThingsBoard": "SubVar",
						"attributeOnDevice": "Root\\.Objects\\.Simulation\\.TOBJ\\.Param\\.SubVar"
					}
				]
			}
		]
	}
}
```

Behaviour description:
1. This creates 2 devices on thingsboard: `Prosys` & `Prosys var with vars`, both have the exact same variable `SubVar`.
2. When creating/editing the shared attribute `SubVar` in the `Prosys` device, the value _does not_ change on Prosys.
3. When creating/editing the shared attribute `SubVar` in the `Prosys var with vars` device, the value _does_ change on Prosys.
I had this problem on other PLCs too, not just simulated ones.
Here is a screenshot of the node structure of Prosys
[image]

samson0v, 2022-12-21:
Hi @SimoneG97, I reproduce your Prosys Server config and OPC-UA Connector config and have the following result:
[image]
[image]
[image]

So it seems that Prosys doesn't allow write value to this variable type.

SimoneG97, 2022-12-21:
I think you haven't set the variable as writable since it is returning you `BadNotWritable`, I only see AccessLevel as `[CurrentRead, HistoryRead]` in the screenshot you provided.  
It also needs `[CurrentWrite]` as I have shown in my previous screenshot.
Just in case I forgot to mention, I was able to reproduce this also on real OPCUA PLCs, so this means it shouldn't be Prosys related.

samson0v, 2022-12-22:
@SimoneG97 ok, I retest and in the result, I can write the value (OPC-UA Connector config is the same as above).
[image]
[image]
[image]

SimoneG97, 2022-12-22:
This behaviour is working ok on my end too (Variable inside an object).
Now try writing a value on a variable inside a variable.
Like this:
[image]
Writing in `Test` doesn't work in the different tests I've made.

samson0v, 2022-12-26:
@SimoneG97, I found a bug, fixing it...

closed 2022-12-28 by imbeacon