thingsboard/thingsboard-gateway #1022: [BUG] Modbus cannot issue rpc about continuous registers

lljy233333, 2022-12-08:
**Describe the bug**
First at all, I read and write individual registers successfully. But when I use the modbus gateway to issue the rpc command to write continuous registers, the program does not report an error, but the device registers do not change.

**Connector name (If bug in the some connector):**
modbus 

**Error traceback (If available):**
1、Go to 'RPC debug terminal'
2、Input ‘setSignalStrength 20’ Then enter
3、The device registers do not change.

**Versions (please complete the following information):**
 - OS Name and Version:Windows 11
 - ThingsBoard IOT Gateway Version:3.2
 - Python version[3.11.0]

lljy233333, 2022-12-12:
{
  "master": {
    "slaves": [
      {
        "host": "192.168.1.222",
        "port": 5020,
        "type": "tcp",
        "method": "socket",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "pollPeriod": 5000,
        "unitId": 1,
        "deviceName": "TH_sensor",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "tag": "batteryLevel",
            "type": "long",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 2
          }
        ],
        "timeseries": [
          {
            "tag": "humidity",
            "type": "long",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 1
          },
          {
            "tag": "temperature",
            "type": "long",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 0
          },
          {
            "tag": "signalStrength",
            "type": "long",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 3
          }
        ],
        "rpc": [
          {
            "tag": "resetTemperature",
            "type": "16int",
            "functionCode": 6,
            "objectsCount": 1,
            "address": 0
          },
          {
            "tag": "setSignalStrength",
            "type": "long",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 3
          },
          {
            "tag": "getSignalStrength",
            "type": "long",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 3
          }
        ]
      }
    ]
  }
}

lljy233333, 2022-12-12:
> 

{
  "master": {
    "slaves": [
      {
        "host": "192.168.1.222",
        "port": 5020,
        "type": "tcp",
        "method": "socket",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "pollPeriod": 5000,
        "unitId": 1,
        "deviceName": "TH_sensor",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "tag": "batteryLevel",
            "type": "long",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 2
          }
        ],
        "timeseries": [
          {
            "tag": "humidity",
            "type": "long",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 1
          },
          {
            "tag": "temperature",
            "type": "long",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 0
          },
          {
            "tag": "signalStrength",
            "type": "long",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 3
          }
        ],
        "rpc": [
          {
            "tag": "resetTemperature",
            "type": "16int",
            "functionCode": 6,
            "objectsCount": 1,
            "address": 0
          },
          {
            "tag": "setSignalStrength",
            "type": "long",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 3
          },
          {
            "tag": "getSignalStrength",
            "type": "long",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 3
          }
        ]
      }
    ]
  }
}

lljy233333, 2022-12-12:
it is my （modbus.json）file.

> Hi @lljy233333, thank you for your request. We will investigate your problem and let you know about the results. Can you, please, send the Modbus Connector configuration file?

it is my （modbus.json）file.

samson0v, 2022-12-21:
@lljy233333, we haven't support type `"type": "long",`, maybe because of it you can't change value.
[image]

closed 2022-12-29 by imbeacon