thingsboard/thingsboard-gateway #1023: [BUG] Can't Write Multiple Registers Using Modbus Connector RPC

helsmore, 2022-12-08:
**Describe the bug**
Using a Modbus connector RPC I don't seem to be able to write to two sequential registers, it fails silently, however writing to the two targetted registers with two individual RPC calls works as expected.

_The detailed description is as follows:_

I have a TB dashboard for the Modbus device with three RPC button widgets on it:
- One for the `unlock` RPC method, payload set to `34209` (i.e. `0x85A1`).
- One for the `reboot` RPC method, payload set to `20736` (i.e. `0x5100`).
- One for the `unlockAndReboot` RPC method, payload set to `2241941760` (i.e. `0x85A15100`)

As you can see by referring to the Modbus connector configuration in the code block below the `unlock` RPC method writes a to a single register address 38, the `reboot` RPC method writes to a single register address 39, whilst the `unlockAndReboot` RPC method writes to registers 38 and 39 together (or at least that's my intention).

The connected Modbus device is setup to reboot itself as soon as `0x85A1` is written to register 38 followed by `0x5100` being written to register 39.

What I am seeing is that if I press the `unlock` RPC dashboard button followed by the `reboot` RPC dashboard button then the Modbus device reboots successfully, however if I push the `unlockAndReboot` RPC dashboard button then it appears that at least one of the registers isn't being written to as the Modbus device does not reboot.

Note I also have various registers being polled every 5 seconds for attributes and time series data, this is also working ok.

Hopefully this is clear but feel free to ask for clarifications, many thanks.

**Connector name (If bug in the some connector):**
Modbus connector

**Error traceback (If available):**
No RPC-related messages shown in logs when RPC button widget is pushed in dashboard, even when connector log level is set to 'DEBUG'.

**Versions (please complete the following information):**
 - OS: Debian 11 Bullseye
 - Thingsboard IoT Gateway version: v3.2 (Docker)
 - Python version: Set by Docker image

Gateway is connected to ThingsBoard Cloud.

Configs:
`~/.tb-gateway/config/modbus_new_1.json`
```
{
  "master": {
    "slaves": [
      {
        "type": "serial",
        "method": "rtu",
        "port": "/dev/ttymxc1",
        "baudrate": 19200,
        "stopbits": 1,
        "bytesize": 8,
        "parity": "E",
        "timeout": 35,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 5000,
        "unitId": 5,
        "deviceName": <REDACTED>,
        "deviceType": <REDACTED>,
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "attributes": [
          {
            "tag": "runtime",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 7
          }
        ],
        "timeseries": [
          {
            "tag": "status",
            "type": "16uint",
            "functionCode": 3,
            "registerCount": 1,
            "address": 9
          }
        ],
        "attributeUpdates": [
          {
            "tag": <REDACTED>,
            "type": "16uint",
            "functionCode": 6,
            "registerCount": 1,
            "address": 30
          }
        ],
        "rpc": [
          {
            "tag": "unlock",
            "type": "16uint",
            "functionCode": 6,
            "registerCount": 1,
            "address": 38
          },
          {
            "tag": "reboot",
            "type": "16uint",
            "functionCode": 6,
            "registerCount": 1,
            "address": 39
          },
          {
            "tag": "unlockAndReboot",
            "type": "32uint",
            "functionCode": 16,
            "registerCount": 2,
            "address": 38
          }
        ]
      }
    ]
  },
  "slave": null
}
```

`~/.tb-gateway/config/tb_gateway.yaml`
```
thingsboard:
  host: <REDACTED>
  port: 8883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: false
    statsSendPeriodInSeconds: 3600
    configuration: statistics.json
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    tls: true
    accessToken: <REDACTED>
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
#  type: sqlite
#  data_file_path: ./data/data.db
#  messages_ttl_check_in_hours: 1
#  messages_ttl_in_days: 7
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
#  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua_asyncio
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Socket TCP/UDP Connector
#    type: socket
#    configuration: socket.json
#
#  -
#    name: XMPP Connector
#    type: xmpp
#    configuration: xmpp.json
#
#  -
#   name: OCPP Connector
#   type: ocpp
#   configuration: ocpp.json
#
# ========= Customization ==========
#
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
#
#  -
#    name: GRPC Connector 1
#    key: auto
#    type: grpc
#    configuration: grpc_connector_1.json
#
  -
    name: Modbus Connector
    type: modbus
    configuration: modbus_new_1.json
    useGRPC: false

```

helsmore, 2022-12-08:
N.B. This sounds similar to https://github.com/thingsboard/thingsboard-gateway/issues/1022 at a glance.

samson0v, 2022-12-26:
Hi @helsmore, trying to fix it...

closed 2022-12-29 by imbeacon